ArduPilot/ardupilot #16661: Tools: OS X Environment Install script broken on Big Sur.

jmachuca77, 2021-02-18:
## Bug report

**Issue details**
The `install-prereqs-mac.sh` script is broken, at least on Big Sur, the following is the output from a fresh install of OS X trying to run it:

```
(master) $ ./install-prereqs-mac.sh 
Checking homebrew...
installing homebrew...
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/universal-darwin20/rbconfig.rb:229: warning: Insecure world writable dir /Users/jaime/Developer/AION/ardupilot_private/Tools/autotest in PATH, mode 040777
Warning: The Ruby Homebrew installer is now deprecated and has been rewritten in
Bash. Please migrate to the following command:
  /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"

==> Checking for `sudo` access (which may request your password).
Password:
==> This script will install:
/usr/local/bin/brew
/usr/local/share/doc/homebrew
/usr/local/share/man/man1/brew.1
/usr/local/share/zsh/site-functions/_brew
/usr/local/etc/bash_completion.d/brew
/usr/local/Homebrew
==> The following existing directories will be made group writable:
/usr/local/bin
/usr/local/share
/usr/local/share/man
/usr/local/share/man/man8
==> The following existing directories will have their owner set to jaime:
/usr/local/bin
/usr/local/share
/usr/local/share/man
/usr/local/share/man/man8
==> The following existing directories will have their group set to admin:
/usr/local/bin
/usr/local/share
/usr/local/share/man
/usr/local/share/man/man8
==> The following new directories will be created:
/usr/local/etc
/usr/local/include
/usr/local/lib
/usr/local/sbin
/usr/local/var
/usr/local/opt
/usr/local/share/zsh
/usr/local/share/zsh/site-functions
/usr/local/var/homebrew
/usr/local/var/homebrew/linked
/usr/local/Cellar
/usr/local/Caskroom
/usr/local/Frameworks

Press RETURN to continue or any other key to abort
==> /usr/bin/sudo /bin/chmod u+rwx /usr/local/bin /usr/local/share /usr/local/share/man /usr/local/share/man/man8
==> /usr/bin/sudo /bin/chmod g+rwx /usr/local/bin /usr/local/share /usr/local/share/man /usr/local/share/man/man8
==> /usr/bin/sudo /usr/sbin/chown jaime /usr/local/bin /usr/local/share /usr/local/share/man /usr/local/share/man/man8
==> /usr/bin/sudo /usr/bin/chgrp admin /usr/local/bin /usr/local/share /usr/local/share/man /usr/local/share/man/man8
==> /usr/bin/sudo /bin/mkdir -p /usr/local/etc /usr/local/include /usr/local/lib /usr/local/sbin /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Frameworks
==> /usr/bin/sudo /bin/chmod g+rwx /usr/local/etc /usr/local/include /usr/local/lib /usr/local/sbin /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Frameworks
==> /usr/bin/sudo /usr/sbin/chown jaime /usr/local/etc /usr/local/include /usr/local/lib /usr/local/sbin /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Frameworks
==> /usr/bin/sudo /usr/bin/chgrp admin /usr/local/etc /usr/local/include /usr/local/lib /usr/local/sbin /usr/local/var /usr/local/opt /usr/local/share/zsh /usr/local/share/zsh/site-functions /usr/local/var/homebrew /usr/local/var/homebrew/linked /usr/local/Cellar /usr/local/Caskroom /usr/local/Frameworks
==> /usr/bin/sudo /bin/mkdir -p /usr/local/Homebrew
==> /usr/bin/sudo /usr/sbin/chown -R jaime:admin /usr/local/Homebrew
==> /usr/bin/sudo /bin/mkdir -p /Users/jaime/Library/Caches/Homebrew
==> /usr/bin/sudo /bin/chmod g+rwx /Users/jaime/Library/Caches/Homebrew
==> /usr/bin/sudo /usr/sbin/chown -R jaime /Users/jaime/Library/Caches/Homebrew
==> Downloading and installing Homebrew...
remote: Enumerating objects: 45, done.
remote: Counting objects: 100% (45/45), done.
[... 855 lines cut ...]
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 2434, in load
    return self.resolve()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/pkg_resources/__init__.py", line 2440, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/Library/Python/2.7/site-packages/pip-21.0.1-py2.7.egg/pip/_internal/cli/main.py", line 60
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
./install-prereqs-mac.sh: line 30: realpath: command not found
usage: dirname path
./install-prereqs-mac.sh: line 31: realpath: command not found
Add /Tools/autotest to your PATH [N/y]?yfinished
```

Break down of issues:
Homebrew formulas broken for:
px4-dev.rb
px4-sim.rb
fastrtps.rb
No formula for gcc-arm-none-eabi

missing commands:
realpath not found

I will try to fix it but might need help testing afterwards.

khancyr, 2021-02-18:
It may worth it switch to python3

jmachuca77, 2021-02-18:
yeah, that's what Im trying to do

jmachuca77, 2021-02-19:
Fixed with #16667
However I can't find a way to force OS X to use python3 by default without disabling the integrity protection. With that enabled /usr/bin/python is simlinked to python2.7 so when running sim_vehilce.py that tries to run using that executable. If the first line in sim_vehicle.py is modified to `#!/usr/bin/env python3` then it works perfectly.
@khancyr @bugobliterator any ideas?

khancyr, 2021-02-19:
That is quite unexpected that they default to python2. 
I think I read somewhere that they made a special python release for the latest mac version.
Anyways, I don't have mac, so I won't be a good helper on this

auturgy, 2021-02-19:
It’s probably worth starting from scratch on this: 90% of that script we don’t need, use, or want.

auturgy, 2021-02-19:
Using in a GitHub Action (Mac is an available environment) as part of CI would go a long way to keeping it in good shape.

bugobliterator, 2021-02-19:
> Fixed with #16667
> However I can't find a way to force OS X to use python3 by default without disabling the integrity protection. With that enabled /usr/bin/python is simlinked to python2.7 so when running sim_vehilce.py that tries to run using that executable. If the first line in sim_vehicle.py is modified to `#!/usr/bin/env python3` then it works perfectly.
> @khancyr @bugobliterator any ideas?
> pyenv is not working for me anymore under Big Sur

Highly recommend using pyenv.

srmainwaring, 2021-02-19:
I've had mixed results with pyenv when trying to build ROS components using catkin and have found the easiest way to have python3 as default is to prefix the path with the brew python installation. This works for ROS and ArduPilot:

```bash
export PATH=/usr/local/opt/python3/libexec/bin:$PATH
```

jmachuca77, 2021-02-19:
@bugobliterator Yeah, I have also used pyenv in the past, but it doesn't work on Big Sur for me, do you have it on Big Sur? The only version of python I seem to be able to install on Big Sur with pyenv is 3.9.1 (@khancyr which is the first version that supports Big Sur) but when I set it as the global version. pyenv reports it as being the version set (`pyenv version`) but when I run `python --version` I still get the system version.

andyp1per, 2021-02-19:
I gave up on pyenv - it didn't support many of the mac features I wanted, but I know I failed to replicate my issues with @bugobliterator so maybe it's just me. I will only be installing native macOS python from python.org - that has consistently given me good results.

bugobliterator, 2021-02-20:
> @bugobliterator Yeah, I have also used pyenv in the past, but it doesn't work on Big Sur for me, do you have it on Big Sur? The only version of python I seem to be able to install on Big Sur with [...]

Hmmm, yes I am using It on Big Sur. Did you add pyenv init to ~/.bash_profile . Very interesting that people are having issues with it. I have been using it for couple years now. Never faced an issue.

jmachuca77, 2021-02-20:
Yeah I added the init and still didn't work, I'll try it again tomorrow. Also I had issues with MAVProxy when I installed it from source so I have to do more investigations

jmachuca77, 2021-03-05:
Ok, I updated the PR with pyenv as well as other changes, I tested it on a fresh install of Big Sur on a parallels VM (because I don't really want to reformat my computer :) )... So if anyone wants to test it, be my guest.... if not I'll bring it up on Mondays dev call and have it merged.

tmoulds, 2021-05-12:
I new to this environment, but working on MacOS Catalina and having this problem, Happy to be a test dummy for your updates. Please let me know how I can help?

IamPete1, 2021-09-12:
Fixed by https://github.com/ArduPilot/ardupilot/pull/17860

closed 2021-09-12 by IamPete1