ArduPilot/ardupilot #16679: Dshot not working Omnibus F4 pro

rac4433, 2021-02-20:
## Bug report

**Issue details**

Cannot connect to Emax Blheli_s ESC through F4 pro FC . The esc functions , the motor runs but I cannot connect and change settings . I have used same esc / computer / drivers / software on Kakute F7 AIO and everything works fine .

There must be something missing in the firmware for the F4 board . No , mission planner is disconnected when I try to connect to esc ( I make sure console is open ) the settings in arduplane are all correct . It just does not connect .
I also don't have any group conflicts and use the correct channel for esc/throttle . I do not have any USB / STM32 driver issues since everything else works fine .

Once again everything works except I can't connect to BLheli with this flight controller when others work fine . The only difference is the flight controller ( 4.0.6) 

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

andyp1per, 2021-02-20:
I think you mean "BLHeli_S passthru" rather than dshot. Unfortunately the passthrough code is very timing sensitive, it works with some ESCs and boards and not others.

rac4433, 2021-02-21:
Hello
Ive lost the ability to connect the esc to BLHeli completely . Both F7 and F4 boards will not connect over two different computers !It just doesn't connect to the flight controller says the error message . I tried using dshot150 instead of dshot600 . no change 
I got it to connect a week ago . I've noticed that with ardupilot , you'll get something working only to come back the next day and it doesn't work anymore ....without anything getting changed , nothing !
I'll probably get rid of the dshot esc and go back to a standard PWM type . I only bought the dshot esc because it was VERY small and fit nice in 600mm mini AR wing .With a fixed wing model and one motor a std esc  will work ( at least it did a few days ago ) . 

I can't even imagine having to setup a more complex multirotor or VTOL using this ardupilot , no way ! I'd rather be subjected to waterboarding !
Thanks for the reply !
RC

andyp1per, 2021-02-21:
I just tried this with an omnibusf4pro and littlebee summer BLHeli_S on 4.1 - worked fine

closed 2021-02-21 by andyp1per

rac4433, 2021-02-21:
Hello
Are you using arduplane or one of the others ? The firmware for arduplane was at 4.0.8 just a few days ago .I just read the wiki where they say only BLHeli_32 is supported ?Just a week ago I was able to connect BLHeli_s esc and get passthrough with kakute f7 

R Champagne