ArduPilot/ardupilot #16681: make: *** No rule to make target `px4-v2'.  Stop.

s-amal, 2021-02-20:
I have windows7 i downloded the ardupilot repository and px4 toolchaine, and when i build a target on px4 eclipse i get this :+1: 

**** Build of configuration Default for project ArduCopter2 ****
make px4-v2 
make: *** No rule to make target `px4-v2'.  Stop.

Any ideas on what might be wrong ?

IamPete1, 2021-02-20:
We now use waf (for the last 3 or 4 years), you will have to setup cygwin on windows seven, instructions here:

https://ardupilot.org/dev/docs/building-setup-windows.html#building-setup-windows

closed 2021-02-20 by IamPete1

peterbarker, 2021-02-21:
On Sat, 20 Feb 2021, s-amal wrote:

> 
> I have windows7 i downloded the ardupilot repository and px4 toolchaine, and when i build a target on px4 eclipse i get this 👍
> 
> **** Build of configuration Default for project ArduCopter2 ****
> make px4-v2
> make: *** No rule to make target `px4-v2'. Stop.
> 
> Any ideas on what might be wrong ?

What instructions are you following to build the code?

s-amal, 2021-03-04:
@peterbarker  i just cloned the repository and followed these instruction to the end    https://ardupilot.org/dev/docs/editing-the-code-with-eclipse.html    i couldnt build any target , i always get that msg

peterbarker, 2021-03-05:
On Thu, 4 Mar 2021, s-amal wrote:

> @peterbarker i just cloned the repository and followed these instruction to the end https://ardupilot.org/dev/docs/editing-the-code-with-eclipse.html i
> couldnt build any target , i always get that msg

I've created a PR to add an out-of-date warning to that page.  I don't use 
eclipse (or Windows)....

https://github.com/ArduPilot/ardupilot_wiki/pull/3367/files