ArduPilot/ardupilot #16685: In ArduPlane, navigation.cpp, calc_gndspeed_undershoot() missing crosswind handling

nagygergely1975, 2021-02-20:
## Bug report
In ArduPlane, navigation.cpp, calc_gndspeed_undershoot() compares non-correlated vectors 

**Issue details**
In ArduPlane, navigation.cpp, calc_gndspeed_undershoot() is comparing GPS track with Yaw / Heading, which is oly valid if there is zero cross-wind (wind is perfectly aligned to Heading - almost never true in real-life). 

_Please describe the problem_
Already lost 3 planes in windy days, so investigated the code. In the function above, If Crosswind-is Non-zero, then GPS track should not be compared to Yaw - instead, it should be compared to Bearing to next WP, not Yaw / Heading. Only a suggestion by a newbie to the Dev program (also a PPL training pilot), so modification is to be considered by more expert Devs. I already submitted my suggested CodeFix as a Git Pull Request, but I am not sure if I have done everything properly as I don't see it as appearing anywhere. Happy if I could contribute.

**Version**
_What version was the issue encountered with_
Latest Plane Master, as of 20/02/2021
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
All fixed-wing types

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk 2.4.8

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2021-02-21:
Are you suggesting that this code caused your crash? Or just that it seemed wrong when you were looking through the code? As far as I can tell it should just cause the vehicle to fly faster than min airspeed.

nagygergely1975, 2021-02-21:
I am not ceratin that all 3 crashes were caused by this (supposed) error / flaw, but is may have contributed significantly, as I have been flying these planes in heavy crosswind-scenarios (meteorological research& measurement). However, I have re-created some of the scenarios with a real-life plane, withour props, carried by car, intentianally getting off course. Here are 2 Vids, the 1st one showing the original calc_gndspeed_undershoot routine, and the 2nd is with my (suggested) fix.

nagygergely1975, 2021-02-21:
First vid - Original calc_gndspeed_undershoot - 
https://www.youtube.com/watch?v=IK1p47G7jG4

nagygergely1975, 2021-02-21:
2nd vid - Same scenario with my suggested Fix, sent in as a Git Push req. id d5c255d60b -
https://www.youtube.com/watch?v=VZ9TF5NUeJk

nagygergely1975, 2021-02-21:
Also not that (I believe) to have discovered another fault in Navigation.cpp call void Plane::setup_turn_angle(void) - It completly appears to be missing the Crosswind compensation when calculating the Desired Turn Angle to the next WP. I am assuming that it might indicate an even deeper, more significant problem! If it is true, then ArduPlane Navigation is simply ignoring Crosswind Compensation - as a training PPL pilot I assume this leads to a "Curvy" Ground Path, which can include Terrain Obstacles, leading to a Crash! I am not 100 % sure of this, but the flaw in setup_turn_angle(void) appears to indicate that. If it is true, it would require an instant fix - I believe - and also an independent Issue  to be opened. I have created a pretty lame, but informal visualization of what I think is happening - if it helps - and I am asking some more experienced Devs to overview if I am right or wrong, because this appears to be a very significant issue. Thank you, and happy if I could contribute! 
[image]

IamPete1, 2021-02-21:
Do you have a log from a successful flight? ideally also in a similar wind, we might be able to spot if something else is wrong. 

The L1 controller deals with winds in navigation, you will always remain on the track between way points. I suspect the majority of these cases wind is ignored simply because the wind estimate is not that accurate, it more conservative to ignore the effect all together than to try and compensate for wind with a poor estimate, that could result in unexpected behavior. Better to neglect the effect and have consistent, if sub optimal in some conditions, results.

nagygergely1975, 2021-02-22:
Hi there, Of course You are right, but only as long as we are discussing
"normal" flight conditions, with relatively low winds, or even wind calm.

However, if You are trying to fly in High- to Severe wind conditions (like
us), pushing the limits of Your vehicle's maximum Airspeed  (using Full
Throttle) Vs. Wind speed, then these effects should not, and (I believe)
must not be ignored.

We, for example, use our Fixed Wing planes for Meteorological Research,
directly & intentionally sending them into Storms of all kinds. Wind speeds
in these conditions can easily exceed some Plane's maximum Airspeed, so
it's actually Vital to handle all these above. (Of course, our planes have
Airspeed sensors).

I hope this explains why I believe these issues should be addressed &
overviewed / fixed.
Thank you!

Peter Hall <notifications@github.com> ezt írta (időpont: 2021. febr. 21.,
V, 22:31):

IamPete1, 2021-02-22:
I think the best path forward would be to come up with a param set and example mission that replicate the issues your seeing IRL in SITL. You can set the wind speed with the SIM_WIND param. I have tested (in SITL) plane with wind in the past and it mostly seems to cope OK. Mission planner SITL will work, or you could compile locally.

tridge, 2021-02-22:
@nagygergely1975 can you please provide logs of your failed flights?
The suggested change doesn't look right to me. It would fail completely when we are circling for example, as the ground speed in the direction of the WP will be zero (we are on tangent of the circle). It will also be wrong when we are starting a turn in a waypoint mission.

robustini, 2021-02-23:
I have never had problems like this, even with delta wings, they sometimes proceeded between waypoints turned forward 45° following the line without any problem, I have many logs to prove this.
@nagygergely1975 I mean like in the picture you posted, ie following the line but rotated by yaw.

closed 2021-08-16 by IamPete1