ArduPilot/ardupilot #16699: Setup_turn_angle(void) is missing Crosswind compensation too

nagygergely1975, 2021-02-21:
## Bug report
It appears that void Plane::setup_turn_angle(void) in Navigation.cpp completly ignores Crosswind calculation

**Issue details**
Plane::setup_turn_angle(void) in Navigation.cpp seems to be using only the Cureent and Next WP bearing when calculating the desired Turn angle, which is a flaw (if true), as Crosswind-compensation should aso be taken into account,

_Please describe the problem_
When investigating another issue ( https://github.com/ArduPilot/ardupilot/issues/16685 ) I believe to have discovered another fault in Navigation.cpp call void Plane::setup_turn_angle(void) - It completly appears to be missing the Crosswind compensation when calculating the Desired Turn Angle to the next WP. I am assuming that it might indicate an even deeper, more significant problem! If it is true, then ArduPlane Navigation is simply ignoring Crosswind Compensation - as a training PPL pilot I assume this leads to a "Curvy" Ground Path, which can include Terrain Obstacles, leading to a Crash! I am not 100 % sure of this, but the flaw in setup_turn_angle(void) appears to indicate that. If it is true, it would require an instant fix - I believe - and also an independent Issue to be opened. I have created a pretty lame, but informal visualization of what I think is happening - if it helps - and I am asking some more experienced Devs to overview if I am right or wrong, because this appears to be a very significant issue. Thank you, and happy if I could contribute!

**Version**
_What version was the issue encountered with_
Latest Arduplane Master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
All airframes

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk 2.4.8
[image]

**Logs**
_Please provide a link to any relevant logs that show the issue_

pompecukor, 2021-02-23:
I don't know much about the code part, but I am quite sure (from many years of experience) that Arduplane would follow shortest distance when flying from point A to point B. Even in crosswind, it would hold a straight line as long as the airplane can manage it.
If this was not the case. It would be impossible for example to use it for aerial mapping, which is one of the most popular commercial/industrial applications.
The plane would crab even, just to keep a direct heading. Perhaps there is something wrong with your setup or platform. In either case a log will help the nice folks here to get to the bottom of you issue.

IamPete1, 2021-08-16:
closing, need a log of the issue to diagnose.

closed 2021-08-16 by IamPete1