ArduPilot/ardupilot #16736: EKF3: becomes unhealthy beyond 38km from origin (EKF2's limit is about 130km)

rmackay9, 2021-02-25:
Some of EKF3's estimates become unusable (by Copter) when the vehicle is about 38km from the origin.  This can be reproduced in SITL by doing the following:

- start copter SITL with speedup=20
- arm in Loiter
- takeoff to 1km
- rc 1 1000 (to fly west at top speed)
- when the vehicle reaches about 35km the EKF will start becoming unhealthy.  Copter's EKF failsafe will trigger and force the vehicle to land

To see the difference vs EKF2 repeat the test after enabling EKF2 (see below) the EKF will become "unhappy" around 130km but not so bad as to trigger the EKF failsafe even beyond 170km.

- param set EK2_ENABLE 1
- param set EK3_ENABLE 0
- param set AHRS_EKF_TYPE 2

EKF3 log: https://www.dropbox.com/s/vcik15awnt4ytd8/master-loiter-far.bin?dl=0
EKF2 log: https://www.dropbox.com/s/y5zsfyz85won6ob/master-loiter-far-ekf2.BIN?dl=0

I wouldn't classify this as a bug but a limitation.  Still, my feeling is that EKF3's 35km limit is too short while EKF2's 100km limit is reasonable.

peterbarker, 2021-02-25:
Would be a *serious* problem for QuadPlane if the same applied in ArduPlane.

rmackay9, 2021-02-25:
@peterbarker I haven't tested quadplane but Rover's EKF failsafe kicked in at about 70km.  I guess the difference is caused by different thresholds in the two vehicles although I haven't investigated much yet.  I expect all vehicles will be affected within this range.

magicrub, 2021-02-25:
@priseborough  Any chance this "limitation" could be extended by using double?

pompecukor, 2021-02-25:
> 
> 
> Would be a _serious_ problem for QuadPlane if the same applied in ArduPlane.

Not sure if this helps, but we have flow ArduPlane with EKF3 enabled only, multiple times to 40 and 50km from home in the past, when testing various radio operating radius ranges. We never got EKF error.

priseborough, 2021-03-01:
> @priseborough Any chance this "limitation" could be extended by using double?

It's too early to say. I've not looked at this issue yet and it will need some time spent on replay to understand the root casue.

priseborough, 2021-05-13:
Preliminary investigation indicates a coupling between the GPS velocity -> EKF and velocity control loop that based on the stepping in GPS velocity 'noise' amplitude with distance from origin, may be a precision issue in the simulated GPS data

[image]

This results in larger and larger velocity innovations until eventually the EKF rejects the GPS and the failsafe activates.

[image]

EKF2 is tuned differently.

I will investigate further

rmackay9, 2021-05-13:
@priseborough

great, thanks for looking into this.  @tridge also thought that it was likely a SITL issue and not a real code issue.

magicrub, 2021-05-13:
I have a 100km log running EKF2 but not sure if I can share it. What can I look for to help?

rmackay9, 2021-06-11:
I've done some real flights with EKF2 and EKF3 and put the four logs here.  https://www.dropbox.com/s/utv9tlvpci5hkba/randy-ekf2-ekf3-far-testing-11Jun2021.zip?dl=0

I tested at distances of 60km and 120km (this is the distance between the EKF origin and the vehicle's actual position) and I didn't notice a difference in performance at least in Loiter mode.   The vehicle was actually slightly difficult to fly well even at 60km because Loiter mode felt like it had points of about 0.5m to 1m wide which it would settle into and it could not be made to stay at locations between these points.  I'm unsure if the limitation is in the EKF or the position controller or Loiter controller.

rmackay9, 2021-07-30:
This is resolved in master as part of the "Extreme ArduPilot" project that greatly increased the maximum range to which AP vehicles can travel.

closed 2021-07-30 by rmackay9