ArduPilot/ardupilot #16744: Copter: DO_MOUNT_CONTROL messages should be ignored if no gimbal

rmackay9, 2021-02-25:
Copter may change the heading of the vehicle if a do-mount-control command is received even if the vehicle has no gimbal (aka AP_Mount) at all.  These messages should be completely ignored if there is no gimbal.

The issue is in Copter's [GCS_MAVLINK_Copter::handle_command_mount()](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/GCS_Mavlink.cpp#L692) method where it checks, "camera_mount.has_pan_control()" and if not it changes copter's "auto_yaw".  Really it should check if there is a configured gimbal first.

YashMaxy, 2021-02-26:
I'd like to fix this issue. Can you please assign it to me?

srirajshukla, 2021-02-26:
How can I check if there is a configured gimbal (AP_Mount)?

srirajshukla, 2021-02-26:
I've not been assigned this issue but can I make a PR for it?

srirajshukla, 2021-02-26:
I've not been assigned this issue, but can I make a PR for it? 

I added this extra if condition before the `camera_mount.has_pan_control()` method : 
```
if (copter.camera_mount.get_mount_type() != copter.camera_mount.MountType::Mount_Type_None)
```

The builds have successfully completed.

rmackay9, 2021-02-26:
I'm very happy to review any PRs that resolve the issue.

@srirajshukla, yes, that fix should work.

Normally we also ask what testing has been done on a PR so it is good to reproduce the problem and then confirm that with the fix it doesn't happen anymore.  I think this testing will probably require using MAVProxy's "msg" or "message" feature (sorry, I'm not sure which is which) to send the DO_MOUNT_CONTROL mavlink message.

If you type "module load msg" and/or "module load message" you can then send any mavlink message from the MAVProxy console.

srirajshukla, 2021-02-27:
I'm starting the SITL in ArduCopter directory with `sim_vehicle.py --map --board`. Both `module load message` and `module load msg` say Loaded module msg/message. But when I pass the command `DO_MOUNT_CONTROL` or `MAV_CMD_DO_GIMBAL_MANAGER_PITCHYAW`, it says Unknown Command. 

Am I missing something, or doing it in the wrong place?

rmackay9, 2021-03-10:
@srirajshukla,

Over on the [mavproxy issues list](https://github.com/ArduPilot/MAVProxy/issues/855) @YashMaxy says that the "module load mount" adds a new mount menu to the right-mouse-button-menu including a "Point here" option that helps demonstrate the problem.

rmackay9, 2021-03-10:
One of the Partners was keen to get this sorted asap so I jumped in and created a PR.  I added you both as Co-authors so thanks for your help!  I still haven't completed testing but it seems good so far.

rmackay9, 2021-03-16:
Merged to master so closing this.

closed 2021-03-16 by rmackay9