ArduPilot/ardupilot #16747: Issues with I2C airspeed on omnibus f4 pro

bobertholomew, 2021-02-26:
I'm running Plane 4.0.9 on an omnibus f4 pro

ARSPD_USE is set to 1 for enable, 
ARSPD_TYPE is set to 1 as I am using the i2c 4525 sensor
ARSPD pin is set to  pix hawk i2c or 65, although I have tried setting it to 1-20 with no luck

I have pull up resistors installed, so I'm not sure where to go from here, please help, thank you.

amilcarlucas, 2021-02-26:
This is probably not a bug, so please post support questions on the forum. This is for confirmed software bugs and feature requests. The forum has a lot more users that can help you out. closing this now.

closed 2021-02-26 by amilcarlucas

bobertholomew, 2021-02-27:
I think it probably is a bug. Everything is set up the way it should be, so it should work, but it doesn't.