ArduPilot/ardupilot #16783: AC407 double stage FS Feature request

roque-canales, 2021-03-02:
Hello, 

it could be interestying to add double stage Radio Failsafe, new FS param for add delay in case of Radiofailsafe, and be able to choose begening FS Strategy between continious last order receved or stop UAV during the FS DELAY.

Radio_FS_DELAY 0-10s
Radio_FS_First action 0 continu last order during Delay or 1 brake and stop drone during this delay

amilcarlucas, 2021-03-02:
This is a new feature, and therefore will not be implemented in ArduCopter 4.0 stable.
If it will be implemented, it will be implemented master.

Please correct the title, and improve the description text. It is too vague.

rmackay9, 2021-03-02:
I thought we already had this on the issues list but I can't find it so I've fixed up the title.

roque-canales, 2021-03-02:
Thank you so much.

Do you think that it could be implemented in the next AC4.1.x ?

amilcarlucas, 2021-03-02:
Github pull-requests are welcome.

roque-canales, 2021-03-02:
can you do this?

rmackay9, 2021-03-02:
@roque-canales,

I'm afraid it is pretty unlikely to be included in 4.1.  SCurves is really the last thing we're working on before starting beta testing.

SergeyBokhantsev, 2021-03-04:
Here is [same request](https://github.com/ArduPilot/ardupilot/issues/9409), I did a draft implementation in my custom build and it's works well. I'm using it from 2018. 
I'm not sure but may be it is easier to do this via scripting nowadays?