ArduPilot/ardupilot #16786: Correct DShot Timings

andyp1per, 2021-03-02:
This is not really a feature request but a documentation of DShot timings for BLHeli that we should use to model our own DShot output on. BLHeli32 and BLHeli_S are incredibly sensitive to timing but always guaranteed to work on Betaflight. Here then are the betaflight timings that we should model as closely as possible

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

andyp1per, 2021-03-02:
Betaflight 4.3.0 w/DShot600 against BLHeli_S 16.7.3 (Channel 0), BLHeli32 32.7 (Channel 1) and no ESC (Channels 4 & 5). 8k gyros 1k PID loops:

[image]

[image]

Pulse-to-pulse is exactly 1ms

andyp1per, 2021-03-02:
[image]

For Channel 0:
ΔT	26.048 µs
Nfalling	16 
Nrising	16 
fmin	600.962 kHz
fmax	603.865 kHz
fmean	601.54 kHz
Tstd	3.312 ns

ΔT	32.552 µs
Nfalling	1 
Nrising	1 

ΔT	23.92 µs
Nfalling	8 
Nrising	8 
fmin	266.525 kHz
fmax	435.54 kHz
fmean	345.713 kHz
Tstd	708.89 ns

For Channel 1:

First pulse is the same then

ΔT	27.616 µs
Nfalling	1 
Nrising	1 

ΔT	26.776 µs
Nfalling	8 
Nrising	8 
fmin	251.004 kHz
fmax	377.644 kHz
fmean	306.802 kHz
Tstd	760.068 ns

andyp1per, 2021-03-02:
At 8KHz PID loop the pulse output is identical but pulse-to-pulse is 125us (i.e. 8Khz)

andyp1per, 2021-03-02:
DShot 300 / 1Khz PIDs:

[image]

Pulse-to-pulse 1ms as before

[image]

Channel 0:
M1:
ΔT	52.112 µs
Nfalling	16 
Nrising	16 
fmin	300.481 kHz
fmax	301.205 kHz
fmean	300.625 kHz
Tstd	3.312 ns

M2:
ΔT	33.8 µs
Nfalling	1 
Nrising	1 

M3:
ΔT	50.048 µs
Nfalling	8 
Nrising	8 
fmin	131.441 kHz
fmax	204.918 kHz
fmean	165.188 kHz
Tstd	1.446 µs

Channel 1:
First pulse the same
M2:
ΔT	30.104 µs
Nfalling	1 
Nrising	1 

M3:
ΔT	53.368 µs
Nfalling	8 
Nrising	8 
fmin	125.502 kHz
fmax	188.253 kHz
fmean	154.104 kHz
Tstd	1.468 µs

andyp1per, 2021-03-02:
DShot 150 / 1Khz PIDs. Interestingly BLHeli_S does not work at this point - it does not return telemetry. Also higher PID rates are disallowed because the DShot pulse will not fit:

[image]

Pulse to pulse still 1ms

[image]

M1:
ΔT	104.424 µs
Nfalling	16 
Nrising	16 
fmin	149.88 kHz
fmax	150.24 kHz
fmean	150.024 kHz
Tstd	4.485 ns

M2:
ΔT	34.416 µs
Nfalling	1 
Nrising	1 

M3:
ΔT	106.488 µs
Nfalling	8 
Nrising	8 
fmin	62.751 kHz
fmax	94.127 kHz
fmean	77.283 kHz
Tstd	2.887 µs

andyp1per, 2021-03-02:
Interestingly DShot1200 is disallowed so I couldn't get any data

andyp1per, 2021-03-02:
500Hz PID loop rate works fine so clearly ESCs do not need 

[image]

giacomo892, 2021-06-04:
Can we close this? @andyp1per

closed 2021-06-04 by andyp1per