ArduPilot/ardupilot #16791: Plane: EKF Altitude Runaway during RTK Fix

Naterater, 2021-03-03:
## Bug report

**Issue details**

Altitude starts to run away prior to arming then constantly resets to 0 every few seconds. After arming, the altitude keeps increasing to ~1000m in maybe 30 seconds with the plane on the ground. Looks to be dangerous to fly, so we disarmed and called it a day.

**Version**
Plane Stable 4.0.9

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Twin motor electric

**Hardware type**
Cube Orange, Here2 (serial/i2c), Emlid Reach M2 (RTK).

**Logs**
https://1drv.ms/u/s!AunBK4rE1ZDCg_MahoKQUbxwQJ8o8w?e=vEWat9

tridge, 2021-03-04:
@Naterater thanks for this report! Could you try and reproduce this with ArduPilot master, and with the following key settings:
 - LOG_DISARMED=1
 - LOG_REPLAY=1
 - 
it needs to be master for the replay to be effective. If we got a replay log of this happening then we could very quickly fix it.

peterbarker, 2021-03-04:
@Naterater as an additional simple check - does this happen with *only* the RTK GPS attached?

peterbarker, 2021-03-04:
.... last time we looked at this it came down to bandwidth; does the same problem occur if you talk to the GPSs at 230400?

tridge, 2021-03-04:
a few notes:
 - First GPS is running at 10Hz, despite the config asking for 5Hz
 - 2nd GPS is running at the requested 5Hz (a u-blox M8)
 - EKF3 buffers will have been sized for reported lag for 1st GPS
 - main problem starts after it switches to the first GPS (the emlid reach)

tridge, 2021-03-04:
I think it is time we remove the buffer length code in EKF3 that depends on reported lag. We should use the max lag always, at least if we have more than 1 GPS configured

tridge, 2021-03-04:
@Naterater I think if you set GPS_DELAY_MS=120 and GPS_DELAY_MS2 to 120 then the problem would not happen
I will develop a PR to fix this, but first I will try to reproduce

Naterater, 2021-03-04:
I'll run some more tests ASAP. We did determine that it appears less likely linked to RTK as we had similar effects after losing RTK early on in the log. It still took about 45 minutes to start diverging though. Logs to follow.

Naterater, 2021-03-04:
And not that it makes too much of a difference, but we did already fix our update rates on our GPS units (10 and 5hz) for the Reach and Here respectively, but the problem still exists.

tridge, 2021-03-04:
> And not that it makes too much of a difference, but we did already fix our update rates on our GPS units (10 and 5hz) for the Reach and Here respectively, but the problem still exists.

it isn't the update rate that matters, it is the delay values

Naterater, 2021-03-04:
Log with the issue and replay enabled as suggested: https://1drv.ms/u/s!AunBK4rE1ZDCg_MeCKOBMYL9W4UIEQ?e=3b6oRq

Note that I can pretty reliably reproduce, and it only shows up after about 45 minutes.

tridge, 2021-03-04:
@Naterater brilliant! replay log replays correctly

Naterater, 2021-03-04:
For what it's worth, I can't seem to repeat the issue on 50Hz loop rate. I'm trying to get you a smaller log, but it's not happening.

tridge, 2021-03-04:
@Naterater did you get a log showing the issue when you had set both GPS_DELAY_MS and GPS_DELAY_MS2 to the same value? (a value of 120 would be suitable)

Naterater, 2021-03-04:
That log is here with each set to 120: https://1drv.ms/u/s!AunBK4rE1ZDCg_MnTod5RWyS_x8Fcg?e=bqQ4da

Naterater, 2021-03-04:
I couldn't get the altitude to show as crazy at 50Hz, so the log while reduced in rate is actually larger because we let it run for longer. https://1drv.ms/u/s!AunBK4rE1ZDCg_MsE_VcwiwTjl2QIw?e=8Psq1l

WickedShell, 2021-03-04:
> a few notes:
> 
>     * First GPS is running at 10Hz, despite the config asking for 5Hz
> 
>     * 2nd GPS is running at the requested 5Hz (a u-blox M8)

The GPS_RATE_MS parameter only applies to uBlox and apparently Allystar NMEA GPS units. Nothing else uses that, and we just cope with whatever we are handed.

priseborough, 2021-03-05:
The EKF is becoming over confident about its accuracy and the state variances collapse just prior to the divergence:

[image]

We have seen this type of behaviour in the past and this log has a number of factors that make it more likely, eg:

1) Lack of movement for extended periods.
2) Faster GPS update rates, eg 10Hz
3) Use of RTK GPS data with small reported error values
4) Use of RTK GPS for height

Reported GPS error levels from the primary GPS in use by both EKF instances including for height is small

[image]
[image]

The next step is to come up with recommended tuning values for 10Hz operation with RTK GPS accuracy, but in the meantime, running the GPS updates at 5Hz would help. It may be necessary with 10Hz operation to limit the observation accuracy used by the horizontal position and height fusion to a minimum of 0.1m instead of the the current hardcoded lower limit of 0.05m.

tridge, 2021-03-05:
>  It may be necessary with 10Hz operation to limit the observation accuracy used by the horizontal position and height fusion to a minimum of 0.1m instead of the the current hardcoded lower limit of 0.05m.

do you mean gpsHgtAccuracy ? That has a lower limit of 1.5 * frontend->_gpsHorizPosNoise, which is 0.75 for this log.

tridge, 2021-03-05:
should we constrain the variances to a min value when inFlight is false?

Naterater, 2021-03-05:
I know that this "divergence" has happened without RTK FIX though. I'm trying to see if I've posted the log yet, but I had one without replay enabled unfortunately. Full disclosure, my recollection is that we did have RTK FIX for about 5-10 minutes during the first 15 minutes of the test, however it lost RTK FIX and degraded to just 3D Fix. 30 minutes later we saw the divergence.

priseborough, 2021-03-05:
The bad news is that neither increasing the minimum observation uncertainty to 0.5m and/or dropping GPS rate to 5Hz was effective on replay. The investigation continues.

priseborough, 2021-03-06:
Correction to my last update. Halving  the GPS measurement rate to 5Hz during replay does fix the divergence, eg:

[image]

The root cause still needs to be investigated and understood.

priseborough, 2021-03-06:
@Naterater Do you have the ability to reduce the output rate of your GPS to 5Hz?

Naterater, 2021-03-06:
@priseborough we do but we don’t. This GPS is also functioning as a precision logger for PPK, and 5Hz reduces our accuracy on turbulent days.

priseborough, 2021-03-08:
See https://github.com/ArduPilot/ardupilot/pull/16842 which fixes this issue on replay.

tridge, 2021-03-08:
@Naterater can you please test this firmware:
http://uav.tridgell.net/tmp/plane-CubeOrange-ekf3-test1.apj
that has the fixes from Pauls PR #16842 
please test with replay enabled. Thanks!

Notorious811, 2021-03-11:
@tridge @priseborough https://discuss.ardupilot.org/t/altitude-estimation-failure/67813/13, it seems like my issue is similar to this, keeping the quadcopter on for longer periods on ground without any movements make the EKF velocity, position and altitude estimations go haywire, should i try to backport this PR https://github.com/ArduPilot/ardupilot/pull/16842 onto the ArduCopter v4.0.4 which i am using currently and do some more tests?

priseborough, 2021-03-26:
Please check that the merge of https://github.com/ArduPilot/ardupilot/pull/16842 has resolved your issue.

Notorious811, 2021-03-27:
Yes, #16842 have resolved this issue for me, I also closed some of my previous issues which was directly related to this issue, seems alright to close this issue as well

rmackay9, 2021-03-29:
Ah, I see @Naterater created this issue so perhaps I closed it prematurely.

I think it will be quite hard to backport this fix to Plane 4.0.x (which is similar to Copter-4.0.x) because there have been so many EKF3 changes since then so it will probably be best to test with "latest" if that's possible.  Of course as you know "latest" hasn't gone through beta testing so there is a higher chance of other issues/bugs.

closed 2021-03-29 by rmackay9