ArduPilot/ardupilot #16802: Move forward/backwards servo output bug

idorutm, 2021-03-03:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

While trying to manouver with my sub, I noticed the forward/backward(reverse) movements were half-powered.
So I looked at the servo_raw tab (photos attached) and saw that I only get a max of 1500 pwm value when doing that movement.
[image]

When moving right/left this issue doesn't occur.
[image]

Can you please refer me to the code-line where this happens? or help me solve this issue?
Thanks in advance!

**Version**
4_2_0 DEV

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[ X ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_ 

BlueROV2/Vectored

**Hardware type**
Pixhawk 2.1 - Black Cube

Williangalvani, 2021-03-03:
I believe it could be related to [this](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_Motors6DOF.cpp#L461).

That said, I'm unable to replicate your issue in SITL.
I have a few questions that could help diagnose the issue (a dataflash log would answer most of them).
What is your setup like?
What flight mode is this?
Do you have any code changes?
Why are your vertical thrusters maxed up?

jmachuca77, 2021-03-03:
This is most likely a settings issue and not a bug in the code. Please post your issue on the user support forums. 

Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)

rmackay9, 2021-03-03:
@Williangalvani,

As @jmachuca77 says this is probably a support call.  How do you feel about us creating a Sub section on our discourse server?  There would be no requirement for you to monitor it of course because I suspect you already have your own support servers on the BlueRov site.

closed 2021-03-04 by rmackay9

Williangalvani, 2021-03-04:
Right. @idorutm feel free to move this on to https://discuss.bluerobotics.com/

@rmackay9 I remember there used to be a Sub category in Ardupilot Discourse that redirected to Bluerobotic's. I guess that got lost somehow...

idorutm, 2021-03-07:
> @Williangalvani,
> 
> As @jmachuca77 says this is probably a support call. How do you feel about us creating a Sub section on our discuss server? There would be no requirement for you to monitor it of course because I suspect you already have your own support servers on the BlueRov site.

is this a support call though? I am not using a bluerov's product...