ArduPilot/ardupilot #16804: Rover: releasing E-Stop may lead to immediate 100% throttle output

rmackay9, 2021-03-04:
During testing of PR https://github.com/ArduPilot/ardupilot/pull/16793 we discovered that E-Stop's status is not being passed to the speed controllers which leads to a few problems:

- GCS output displays 100% throttle when E-Stop is engaged
- Speed controller's I-term builds up (as it fights to try and speed up the vehicle but there is no response because the wheels won't move)
- throttle output may jump to 100% as soon as E-Stop is released

[image]

This can be reproduced in SITL by doing the following

- set speed controller parameters so the vehicle requires a lot of I-term:
  -  param set CRUISE_THROTTLE 100
  -  param set CRUISE_SPEED 5
  -  param set ATC_SPEED_P 0
- param set RC9_OPTION 31 (to enable E-Stop on channel 9)
- param set GCS_PID_MASK 2 (so that PID outputs are sent to GCS)
- graph VFR_HUD.throttle (to view reported throttle output)
- graph SERVO_OUTPUT_RAW.servo3_raw (to view actual throttle output)
- graph PID_TUNING.I (to view speed controller's I-term)
- create a simple auto mission (using "wp editor") with a long straight segment
- arm in Auto
- rc 9 2000 (to engage E-stop and stop the motors)
- rc 9 1000 (to disengage E-stop to allow the motors to spin again)

To resolve this issue:

- we should probably stop calling get_throttle_out_stop or get_throttle_out_speed while E-Stop is active.  If neither of these are called for at least 0.2sec then I think they will handle their own reset
- the _desired_speed (wherever that is held) should probably be reset to zero or the current speed?  We need to be on the lookout for massive changes in the throttle output when users disengage E-stop while the vehicle is still moving.  If there is an issue this could be considered out-of-scope because the same issue may exist if the vehicle switches from Manual to Acro while moving quickly.
- the E-Stop handling could either be inside the vehicle code or inside AR_AttitudeControl.  I'm not a big fan of adding a dependency on SRV_Channel to AR_AttitudeControl but I don't have a really strong view on this.

mikerob, 2021-04-07:
I'm facing a similar potential problem with ICE support for Rover.  I'm concerned that if the engine dies at some point during an AUTO mission and you (or the system) triggers an engine restart, that the I term will have built up and you'll get a throttle jump.

AR_AttitudeControl has a "Relax" function:

`    // relax I terms of throttle and steering controllers
    void relax_I();`

but it relaxes BOTH steering and throttle. One fix would seem to be a publically accessible function that just relaxes the throttle term, then both your E-stop code and my ICE code could call it.

rmackay9, 2021-04-07:
@mikerob,

yes, loss of power from the ICE engine would lead to I-term build-up.  I guess another solution would be to provide feedback from the engine and then use that to limit the controller's I-term build-up.

One small complication is that skid-steering vehicles can get I-term build-up in the steering controlling which is probably why the "relax_I" resets them both...

mikerob, 2021-04-11:
It seems like the ICEngine::update function would be the place to do the check. Even if you don't have RPM feedback, it seems like any time you start the engine, it would be sensible to reset the I term, just to be safe.

How can you access the attitude controller from within the ICE code?  Would you instead do it in GCS_Mavlink and manually intercept an engine command to reset the I term?  That would still need a specific public function to reset just the throttle I.