ArduPilot/ardupilot #16806: Copter/Sub/Plane/Rover: should we send message_item_reached message in Guided?

rmackay9, 2021-03-04:
PR https://github.com/ArduPilot/ardupilot/pull/16796 brought to light the small issue that our vehicles are inconsistent in whether they send message_item_reached messages when they reach a Guided mode target.  Copter and Sub do not.  Plane and Rover do.

This question was raised by @jmachuca77

amilcarlucas, 2021-03-04:
Yes, adding support in Copter and Sub would make a lot of users happy :)

jmachuca77, 2021-03-04:
I believe for guided mode it is important to send that message in all vehicles. Its the only feedback a user has to know that the target they sent was actually reached. A companion computer running a script will want to know when it should send the next target for example. 
In the case of auto mode, I don't think its as critical because AP is in control and will send the next waypoint automatically. But for consistency I would also add it. That way again, if a companion computer is monitoring the progress of a mission it could potentially use those messages to know when AP reached the target.

amilcarlucas, 2021-03-04:
Can it be refractored into AP_Vehicle ?

PR #16816 (2021-03-09): Copter: send mission_item_reached in guided
fix #16806, most code copied from #16796 (thanks @rmackay9)

tested on mro pixracer pro, indoor with motion capture system
closed 2021-03-09 by rmackay9