ArduPilot/ardupilot #16814: Short term GPS glitch can cause long term EKF3 accel bias and attitude error

tridge, 2021-03-04:
We have a bench test log showing EKF3 producing a long term 5 degree roll error, apparently due to a quite short (few second) GPS glitch. It is not a replay log.
[image]
[image]
[image]
[image]
the IMU data is steady, board was not moving in bench test.
The key concern is the persistence of the roll error from the accel bias estimate after such a short GPS event.
I have requested that @Hwurzburg try to reproduce with LOG_REPLAY=1
log is here:
http://uav.tridgell.net/EKF/ekf3-roll-offset-gps1.bin
ping @priseborough

tridge, 2021-03-04:
part of the problem may be the degree that we trust the u-blox gps velocity accuracy data. With no movement in a bench test the EKF is starved of information for bias updates except when the GPS outputs false data.
So I suspect this is really just a bench test artefact and not of concern for flight

priseborough, 2021-03-05:
This is expected behaviour. The biases will re-learn correctly and restore correct roll/pitch when vehicle starts manoeuvring with GPS enabled.

closed 2021-03-05 by priseborough