ArduPilot/ardupilot #16834: Logging Failure if "LOGS" Folder Not Present

SUI-Eric, 2021-03-07:
## Bug report

**Issue details**

If the LOGS folder is not present on an SD card, Ardupilot seems to have lost the ability to create that directory. This results in "Logging not started" pre-arm errors if using a new SD card or if the LOGS folder has been destroyed for any reason.

**Version**
4.0.6 & 4.0.7 (tested so far)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Cube Orange & Cube Blue (F4)

**Logs**
Unavailable by definition. However, this issue is easy to reproduce by the below process:

1. Load a Cube with AC 4.0.6 or 4.0.7
2. Insert a blank SD card or manually delete the LOGS folder from a previously used SD card
3. Ensure logging is part of the arming checks
3. Attempt to arm

amilcarlucas, 2021-03-08:
I think tridge fixed this in master. @SUI-Eric can you test that?

Should the fix be backported? I do not think it is that important to grant a new release.

SUI-Eric, 2021-03-08:
@amilcarlucas I just grabbed the latest build of Master for Copter, and still experience "PreArm: Logging failed" if the SD card doesn't already have the LOGS folder present.

I do think it'd be important to backport the fix when it occurs, as any user who's got a new FC (or just a new SD card) and loads stable will experience it and not know why unless they check here.

rmackay9, 2021-03-09:
I was unable to reproduce this issue using Copter-4.0.7 on a CubeBlack or a CubeOrange.  I'm not saying that there's isn't a problem of course, just saying that I couldn't reproduce it.

SUI-Eric, 2021-03-09:
Interesting. I did some more testing and found that I have one Cube Black (out of four tested so far) that does not seem to be affected and creates the folder each time. The other three reliably reproduce this issue across several SD cards. 

I'll try a Matek board or something to try and determine whether this happens to somehow be a Cube problem instead of a bug.

amilcarlucas, 2021-03-09:
Same bootloader and FW version?

WangLouis, 2021-07-06:
Version 4.1.0  Copter Pixhawk4 mini using a new SD card don't create  LOGS folder.
pre-arm error massage is logging failed.

but using Version 4.0.3 CubeBlack can create  LOGS folder.

IamPete1, 2021-08-16:
related to https://github.com/ArduPilot/ardupilot/issues/16103 ?

IamPete1, 2021-11-16:
Closed by https://github.com/ArduPilot/ardupilot/pull/19267 and https://github.com/ArduPilot/ardupilot/pull/19271

closed 2021-11-16 by IamPete1