ArduPilot/ardupilot #16844: EKF3: Aiding and forced reset loop

stephendade, 2021-03-08:
## Bug report

There appears to be an issue with the EKF constantly resetting on the Rover firmware-

With GPS and default parameters, I get the following text continuously (1 per 2 seconds):
```
APM: EKF3 IMU1 forced reset
APM: EKF3 IMU1 initialised
APM: EKF3 IMU1 initial yaw alignment complete
APM: EKF3 IMU1 tilt alignment complete
```

Without a GPS or compass, I get the following text continuously on the console (a few times per second):
```
APM: EKF3 IMU0 stopped aiding
APM: EKF3 IMU0 fusing odometry
APM: AHRS: DCM active
APM: EKF3 IMU0 started relative aiding
APM: AHRS: EKF3 active
```

I've tested down the commit tree and this first appeared somewhere between 9228925bea78c986dfc1266d4a6b9d4814f5d0bb and 6d6ec44d5982c06a6774296aab8ca732f1b9d104. Can't narrow it down any further due to build errors (due to an EKF2 issue. Unrelated?)

**Version**
Master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X ] Rover
[  ] Submarine

**Hardware type**
Pixhawk1-1M. I've tested several of these boards just in case it was a hardware issue.

priseborough, 2021-03-08:
Please supply  a log that shows the fault generated with 

LOG_DISARMED 1
LOG_REPLAY 1

TXS

stephendade, 2021-03-09:
Here's the log:
[00000004.zip](https://github.com/ArduPilot/ardupilot/files/6107724/00000004.zip)

(There was no GPS connected for this test)

priseborough, 2021-03-13:
@stephendade The reason for the repeated reset of the EKF for the second IMU is an excessive Z accel bias that is resulting in large and sustained vertical position innovations, eg:

[image]

[image]

[image]

Please recalibrate your accelerometers and re-test.

stephendade, 2021-03-14:
@priseborough: I recalibrated the accelerometers and there's no more resets.

Thanks!

closed 2021-03-14 by stephendade

xcl010, 2021-06-06:
> ## Bug report
> There appears to be an issue with the EKF constantly resetting on the Rover firmware-
> 
> With GPS and default parameters, I get the following text continuously (1 per 2 seconds):
> 
> ```
> APM: EKF3 IMU1 forced reset
> APM: EKF3 IMU1 initialised
> APM: EKF3 IMU1 initial yaw alignment complete
> APM: EKF3 IMU1 tilt alignment complete
> ```
> 
> Without a GPS or compass, I get the following text continuously on the console (a few times per second):
> 
> ```
> APM: EKF3 IMU0 stopped aiding
> APM: EKF3 IMU0 fusing odometry
> APM: AHRS: DCM active
> APM: EKF3 IMU0 started relative aiding
> APM: AHRS: EKF3 active
> ```
> 
> I've tested down the commit tree and this first appeared somewhere between [9228925](https://github.com/ArduPilot/ardupilot/commit/9228925bea78c986dfc1266d4a6b9d4814f5d0bb) and [6d6ec44](https://github.com/ArduPilot/ardupilot/commit/6d6ec44d5982c06a6774296aab8ca732f1b9d104). Can't narrow it down any further due to build errors (due to an EKF2 issue. Unrelated?)
> 
> **Version**
> Master
> 
> **Platform**
> [ ] All
> [ ] AntennaTracker
> [ ] Copter
> [ ] Plane
> [ X ] Rover
> [ ] Submarine
> 
> **Hardware type**
> Pixhawk1-1M. I've tested several of these boards just in case it was a hardware issue.

https://github.com/ArduPilot/ardupilot/issues/15916#issuecomment-855377469

priseborough, 2021-06-06:
@xcl010 Please raise an issue for this and provide a log that demonstrates the issue obtained using LOG_DISARMED=1 and LOG_REPLAY=1

rmackay9, 2021-06-06:
@xcl010,

Another option is to create a new topic in the Rover-4.1.0-beta testing category here. https://discuss.ardupilot.org/c/ardurover/rover-41/158