ArduPilot/ardupilot #16855: Arduplane 4.1 VTX option

stevereed66, 2021-03-09:
[2021-03-09 09-00-50.zip](https://github.com/ArduPilot/ardupilot/files/6107405/2021-03-09.09-00-50.zip)

## Bug report

**Issue details**
Constant VTX messages power level Freq etc.
_Please describe the problem_
After initial load all working as expected, I think 1 vtx message at start.
turn on vtx --- constant messages.
turn off vtx still constant vtx messages

options
start vtx_enable=0.

set up vtx option for testing
vtx_enable=1 and appropriate parms all works but constant VTX messages

now set 
vtx_enable=1
vtx_band=0
vtx_channel=0
vtx_max_power=0
vtx_frequency=0
vtx__power=0
vtx_options=0
RCn_OPTION = 94  not set
Set SERIALn_PROTOCOL = 37 not set

still messages

reboot still messages

vtx_enable=0

reboot

still constant vtx message in mission planer  message window osd messages and yappu telemetary.

Have attached tlog (zip)
[2021-03-09 09-00-50.zip](https://github.com/ArduPilot/ardupilot/files/6107418/2021-03-09.09-00-50.zip)

**Version**
_What version was the issue encountered with_
ver 4.1 dev latest
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Nano tallon
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Matek  f765
**Logs**
_Please provide a link to any relevant logs that show the issue_

andyp1per, 2021-03-09:
@stevereed66 what VTX are you using. Can you post your parameter configuration?
Are you using CRSF?

stevereed66, 2021-03-09:
Thanks for quick response

CRSF
TBS UNIFY Pro 5v  V3

attached parms

Just realised while dumping parms 

Boot and connect to mission planner ---  no radio so in failsafe
no vtx message
Turn on radio failsafe off vtx messages start
turn off radio -- in faisafe
vtx message stop .

Flight mode = 11
09/03/2021 10:03:12 : Failsafe. Long event on: type=2/reason=3
09/03/2021 10:03:07 : Flight mode = 1
09/03/2021 10:03:07 : Failsafe. Short event on: type=1/reason=3
09/03/2021 10:03:07 : Throttle failsafe on
09/03/2021 10:03:06 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:03:06 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:03:03 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:03:03 : Relay1 HIGH
09/03/2021 10:03:01 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:03:01 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:58 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:57 : Relay1 LOW
09/03/2021 10:02:56 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:56 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:53 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:51 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:51 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:49 : CRSF: RF mode 2, rate is 156Hz
09/03/2021 10:02:48 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:46 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:46 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:44 : CRSF: custom telem init done, fw 4.08
09/03/2021 10:02:44 : CRSF: RF mode 2, rate is 103Hz
09/03/2021 10:02:43 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:41 : VTX: Freq: 5865MHz, Power: 0mw, Band: 1, Chan: 1
09/03/2021 10:02:39 : Failsafe. Long event off: reason=3
09/03/2021 10:02:39 : Relay1 HIGH
09/03/2021 10:02:39 : Throttle failsafe off
09/03/2021 10:02:39 : RCInput: decoding CRSF
09/03/2021 10:02:39 : CRSF: requesting RX device info
09/03/2021 10:02:39 : CRSF: RF mode 1, rate is 1Hz
09/03/2021 10:02:18 : AHRS: EKF2 active

09/03/2021 10:02:18 : EKF2 IMU1 is using GPS
09/03/2021 10:02:18 : EKF2 IMU0 is using GPS
09/03/2021 10:02:17 : EKF2 IMU1 origin set
09/03/2021 10:02:17 : EKF2 IMU0 origin set
09/03/2021 10:01:05 : EKF2 IMU0 tilt alignment complete
09/03/2021 10:01:00 : EKF2 IMU1 initial yaw alignment complete
09/03/2021 10:01:00 : EKF2 IMU0 initial yaw alignment complete
09/03/2021 10:00:59 : GPS: u-blox 1 saving config
09/03/2021 10:00:45 : u-blox 1 HW: 00080000 SW: ROM CORE 3.01 (107888)
09/03/2021 10:00:38 : IMU0: fast sampling enabled 8.0kHz/2.0kHz
09/03/2021 10:00:38 : RCOut: PWM:1-4 DS600:5-6 PWM:7-12 NeoP:13
09/03/2021 10:00:38 : MatekF765-Win 00280039 34385106 3431353
09/03/2021 10:00:38 : ChibiOS: a14aa6b0
09/03/2021 10:00:38 : ArduPlane V4.1.0dev (4abae233)
09/03/2021 10:00:37 : IMU0: fast sampling enabled 8.0kHz/2.0kHz
09/03/2021 10:00:37 : RCOut: PWM:1-4 DS600:5-6 PWM:7-12 NeoP:13
09/03/2021 10:00:37 : MatekF765-Win 00280039 34385106 3431353
09/03/2021 10:00:37 : ChibiOS: a14aa6b0
09/03/2021 10:00:37 : ArduPlane V4.1.0dev (4abae233)

andyp1per, 2021-03-09:
So for CRSF you need to turn off "my VTX" in the TX

andyp1per, 2021-03-09:
So if you want to configure the VTX from your TX using the crossfire app (or myVTX) them you should not set VTX_ENABLE. VTX_ENABLE is only if you want AP to control the VTX settings.

stevereed66, 2021-03-09:
Vtx enable should be 0 disabled. Is that not what's in my parm file.Steve

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/ArduPilot/ardupilot/issues/16855#issuecomment-793752621",
"url": "https://github.com/ArduPilot/ardupilot/issues/16855#issuecomment-793752621",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

andyp1per, 2021-03-09:
There don't appear to be any attached params - you probably need to use the webUI to attach rather than e-mail

stevereed66, 2021-03-09:
[4.1latestdump.zip](https://github.com/ArduPilot/ardupilot/files/6108697/4.1latestdump.zip)

andyp1per, 2021-03-09:
ok, I see - you are getting spammed even though the VTX code should be disabled. I have seen this before with @yaapu 's passthrough enabled - can you try disabling passthrough and see if you still get the spam?

stevereed66, 2021-03-09:
Yes will get back.I also get more messages in yaapu eg no receiver. Alt varience compass varience which are not in tlog or data flash.   Nothing in logs to suggest the above.  I am waiting on yaapu to reply.  Thanks great workSteve

stevereed66, 2021-03-09:
Ok uploaded

stevereed66, 2021-03-09:
I  have another set up, hope it helps
Tilt rotor quadplane
Matek h743
same vtx and crossfire and rc_options=288

vtx_options have never been set to 1 always zero.
Do not get messages

It seems I only had messages on f765 after setiing vtx_option=1, testing then back to 0, the messages never stopped while TX is on.

stevereed66, 2021-03-09:
just thought I would set rc_options on f765 to 32, default. Still have the vtx constant message.
I can imagine you looking at code with big cup of coffee.
IF you would me to change any parameters/wiring please let me know, raining all day tomorrow

andyp1per, 2021-03-09:
[arduplane.zip](https://github.com/ArduPilot/ardupilot/files/6110973/arduplane.zip)

Can you try the attached firmware to see if it helps? You will need to unzip it

stevereed66, 2021-03-09:
rc_options back to 356
reboot
All vtx messages now gone
tx on or off

yaapu telemetry working

Getting occasional crt No RC receiver   in Yaapu Messages.  This is why I set Ignore RC fai safe bit to see if this will stop message.

Have attached Tlog

So you have fixed VTX message problem really quickly thanks

Do you want me to turn on vtx_options test and turn off.

Am I able to fly with this version ?

Here is mission planner message window
[2021-03-09 19-56-06.zip](https://github.com/ArduPilot/ardupilot/files/6111149/2021-03-09.19-56-06.zip)

09/03/2021 20:00:58 : CRSF: RF mode 2, rate is 171Hz
09/03/2021 20:00:41 : CRSF: RF mode 2, rate is 145Hz
09/03/2021 20:00:36 : CRSF: RF mode 1, rate is 165Hz
09/03/2021 20:00:11 : CRSF: RF mode 2, rate is 159Hz
09/03/2021 20:00:05 : Failsafe. Short event off: reason=3
09/03/2021 20:00:05 : Throttle failsafe off
09/03/2021 20:00:05 : Flight mode = 11
09/03/2021 20:00:05 : Failsafe. Short event on: type=1/reason=3
09/03/2021 20:00:05 : Throttle failsafe on
09/03/2021 20:00:02 : Failsafe. Long event off: reason=3
09/03/2021 20:00:02 : Throttle failsafe off
09/03/2021 20:00:02 : CRSF: RF mode 2, rate is 131Hz
09/03/2021 19:59:55 : Flight mode = 1
09/03/2021 19:59:55 : Failsafe. Short event on: type=1/reason=3
09/03/2021 19:59:55 : Throttle failsafe on
09/03/2021 19:59:50 : CRSF: RF mode 1, rate is 171Hz
09/03/2021 19:59:37 : CRSF: RF mode 2, rate is 164Hz
09/03/2021 19:59:32 : CRSF: RF mode 1, rate is 171Hz
09/03/2021 19:59:22 : AHRS: EKF2 active
09/03/2021 19:59:22 : EKF2 IMU1 is using GPS
09/03/2021 19:59:22 : EKF2 IMU0 is using GPS
09/03/2021 19:59:21 : EKF2 IMU1 origin set
09/03/2021 19:59:21 : EKF2 IMU0 origin set
09/03/2021 19:58:43 : CRSF: RF mode 2, rate is 154Hz
09/03/2021 19:58:38 : CRSF: RF mode 1, rate is 171Hz
09/03/2021 19:57:52 : CRSF: custom telem init done, fw 4.08
09/03/2021 19:57:29 : CRSF: RF mode 2, rate is 151Hz
09/03/2021 19:57:24 : CRSF: RF mode 1, rate is 167Hz
09/03/2021 19:56:18 : EKF2 IMU1 initial yaw alignment complete
09/03/2021 19:56:18 : EKF2 IMU0 initial yaw alignment complete
09/03/2021 19:56:16 : CRSF: RF mode 2, rate is 154Hz
09/03/2021 19:56:11 : CRSF: RF mode 1, rate is 170Hz
09/03/2021 19:56:10 : IMU0: fast sampling enabled 8.0kHz/2.0kHz
09/03/2021 19:56:10 : RCOut: PWM:1-4 DS600:5-6 PWM:7-12 NeoP:13
09/03/2021 19:56:10 : MatekF765-Win 00280039 34385106 3431353
09/03/2021 19:56:10 : ChibiOS: a14aa6b0
09/03/2021 19:56:10 : ArduPlane V4.1.0dev (64352312)
09/03/2021 19:56:10 : IMU0: fast sampling enabled 8.0kHz/2.0kHz
09/03/2021 19:56:10 : RCOut: PWM:1-4 DS600:5-6 PWM:7-12 NeoP:13
09/03/2021 19:56:10 : MatekF765-Win 00280039 34385106 3431353
09/03/2021 19:56:10 : ChibiOS: a14aa6b0
09/03/2021 19:56:10 : ArduPlane V4.1.0dev (64352312)

andyp1per, 2021-03-09:
Yes please also try with VTX_OPTIONS on and check that things are still working - you will need to disable "myVTX" on the transmitter

stevereed66, 2021-03-10:
re ( you will need to disable "myVTX" on the transmitter) Sorry how do I do that, what does this do.
 Will try VTX_OPTIONS tomorrow

andyp1per, 2021-03-10:
In the crossfire telemetry script there is an option for doing this for the transmitter

stevereed66, 2021-03-10:
set vtx_options :

VTX_BAND	4
VTX_CHANNEL	1
VTX_ENABLE	1
VTX_FREQ	5695
VTX_MAX_POWER	4    or 800    note sure units  message says power 0
VTX_OPTIONS	3              
VTX_POWER	1    or 100    

Back to constant messages only when TX on

10/03/2021 11:07:14 : Flight mode = 11
10/03/2021 11:07:14 : Failsafe. Long event on: type=2/reason=3
10/03/2021 11:07:09 : Flight mode = 11
10/03/2021 11:07:09 : Failsafe. Short event on: type=1/reason=3
10/03/2021 11:07:09 : Throttle failsafe on
10/03/2021 11:07:06 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:07:04 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:07:01 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:06:59 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:06:56 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:06:54 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:06:49 : CRSF: RF mode 2, rate is 169Hz
10/03/2021 11:06:49 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:06:46 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:06:44 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:06:42 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:06:42 : CRSF: RF mode 2, rate is 143Hz
10/03/2021 11:06:39 : VTX: Freq: 5695MHz, Power: 0mw, Band: 5, Chan: 2
10/03/2021 11:06:37 : Failsafe. Long event off: reason=3

VTX_ENABLE 0

vtx messages stop

stevereed66, 2021-03-10:
XF Micro TX
MY VTX
ACTIVE NO   Has never been switched on

andyp1per, 2021-03-10:
Ok, so I guess confusing why you still get the spam then with VTX_ENABLE on - are you trying to control the VTX using anything else?

stevereed66, 2021-03-10:
OK Have got messages to stop with VTX_Enable=1

controlling VTX with opentx Nano RX APP
Set XF NANO Output 4 to Smart Audio  in OPenTX App
wiring from RX 4 to Unify Smart Audio
Constant VTX messages

Set XF NANO  to CH5 in OPenTX App 
messages stop.

put back 
Set XF NANO Output 4 to Smart Audio  in OPenTX App
Messages start again.

So I guess this is expected  behaviour.

So I can text Ardu VTX  I need to Know

What power setting values for Unify 800 mw

VTX_MAX_POWER
VTX_POWER

What bitmask for  pit mode unarmed,  full power armed, guess  I also need unlock
VTX_OPTIONS 

VTX_OPTIONS - @Bitmask: 0:Pitmode:1:Pitmode until armed,2:Pitmode when disarmed,3:Unlocked
VTX_POWER - @Description: Video Transmitter Power Level. Different VTXs support different power levels, the power level chosen will be rounded down to the nearest supported power level

 OSD parameter support, how is this used.

Thanks

Steve

andyp1per, 2021-03-10:
Hey you should have XF NANO set to CRSF, SA is definitely wrong.

The rest will be in the docs when I get around to writing them up 😄

stevereed66, 2021-03-10:
Output 1 is crsf TX
            2 is crsf RX
Would not work else
Out 4 is set to SA when using TBS to control VTX, which causes messages.

Will test when you have published docs

IamPete1, 2021-08-16:
@andyp1per sounds like this issue is resolved?

andyp1per, 2021-08-16:
Should be

closed 2021-08-16 by andyp1per