ArduPilot/ardupilot #16905: Copter: KDECAN throttle output is wrong.

KDEDirect, 2021-03-15:
**Issue Details**

A higher throttle value is required to spin the motors when using KDECAN ESCs. Sending 0% throttle sends CAN bus data of 400 instead of 1000 so the motors don't start spinning until above 30% throttle. As a temporary solution we have been recommending customers change the Output PWM Min to 1500 and Output PWM Max to 1900 to achieve an output range of about 1000-2000. This issue is also referenced here: https://github.com/ArduPilot/ardupilot/issues/15072

**Version**
firmware built from master

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad, Hex, Octo

**Hardware type**
Pixhawk, Cube, DrotekP3Pro, etc.

**Solution**
The SRV_Channel::get_output_norm() results in incorrect output. It should be replaced with SRV_Channel::get_motor_function() in the KDECAN library. The Toshiba library does this and it works properly.

Reference:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_KDECAN/AP_KDECAN.cpp#L568
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_ToshibaCAN/AP_ToshibaCAN.cpp#L436

We tested the following code and it produced a correct output range for CAN bus:

[image]

With the code above, throttle 0% sends 1000 and throttle 100% sends 2000.

magicrub, 2021-03-15:
Has @KDEDirect considered implementing the UAVCAN protocol? I'm sure developers from Ardupilot would love to help you any way they can to make that happen on your end. Please consider this an open invite from the developer community - we're here to help! It's win-win.

KDEDirect, 2021-03-15:
Thank you for the invite and recommendation. At this time KDE Direct is pursuing R&D that requires the use of KDECAN.

rmackay9, 2021-04-12:
@KDEDirect,

Hi there, I think I've got a fix for this issue which could be included in the upcoming Copter-4.1 release. The PR is here https://github.com/ArduPilot/ardupilot/pull/17149 but if you'd like to test I could create a binary for you if you tell me which autopilot you are using.

rmackay9, 2021-04-19:
Fixed in master so closing.  This will go out for user testing with Copter-4.1.0-beta2 within a few weeks.

closed 2021-04-19 by rmackay9

KDEDirect, 2021-04-23:
Thanks so much for fixing this. I tested the master branch today and it works great. I will update the instructions on our website to specify setting CAN_Dx_PROTOCOL to "8" on Copter-4.1.0-beta2 and above.

rmackay9, 2021-04-24:
@KDEDirect,

No problem.  by the way, while setting up my KDECAN ESCs I struggled to get the wiring correct.  It seems that the pins are all reversed (1<->4 and 2<->3).  Is there a chance that KDE could fix the cables so that at least one end matches the standard?  It's annoying for users to have to manually pull out the pins on one cable and swap them around.  To be clear, you'd only need to do one cable but then label it somehow so that users know this is the one used to connect the 1st ESC to the autopilot.

We will update the wiki to specify this but.. it would be nice if users didn't have to go through this.

rmackay9, 2021-04-26:
I've fixed up the wiki now as well https://ardupilot.org/copter/docs/common-kde-can-escs.html.  It might take 30min after this post for the changes to appear but now I think users should be able to get these ESCs working.