ArduPilot/ardupilot #16940: Waf Configure Fails  (FileNotFoundError: [Errno 2] No such file or directory: '/dev/nul')

b-madhukar, 2021-03-18:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

b-madhukar, 2021-03-18:
I have tried with /dev/null and /dev/nul on windows 10 platform, both dint work

b-madhukar, 2021-03-20:
Can anyone please check this issue , am stuck here

IamPete1, 2021-08-16:
were you able to resolve this?

dengtong2018, 2021-12-28:
were you has to resolve this?

hendjoshsr71, 2021-12-28:
> Did you solve the problem?

Unless you need to compile SITL binaries for Mission Planner we suggest using WSL2 for a windows development environment.
https://ardupilot.org/dev/docs/building-setup-windows10.html