ArduPilot/ardupilot #16949: Parameters are not persistant on reboot

ademczuk, 2021-03-20:
## Bug report

**Issue details**
 Firmware does not retain any parameter setting on https://firmware.ardupilot.org/Rover/latest/MatekH743/ardurover_with_bl.hex for this flight controller. This issue does not occur on the latest arducopter or ArduRover V4.1.0-dev (37cf09a8) from 2020-10-18.
I tested this on 19 March 2021
Host System: Windows 10
No peripherals attached, SD card only
I noticed that two com ports were available, which was unusual as only one should appear. 

I requested that another person try this on their same flight controller and they also had the same results:
Date:            19 March 2021
Host system:        macOS 10.15.7
Flight controller:    Mateksys H743-WING

Test 1: known good

No peripherals attached, SD card only

Firmware:        ArduRover V4.1.0-dev (37cf09a8)
Flash:            Install with ST32CubeProgrammer after full chip erase
SD card:            SanDisk Ultra microSDHC UHS-I Card, 16 GB, 98 MB/s

GSC:            Mission Planner (running on Windows 10 VMware VM)

Config > Full Parameter List > SCR_ENABLE = 1 (then reload params and restart)
Config > MAVFtp > APM > scripts > hello_world.lua

A. USB connection only, no battery

Data > Messages:    hello world script running correctly

B. USB and 14.4V NiMH battery

Data > Messages:    hello world script running correctly
Test 2: install newer version of ArduPilot

Firmware:        ArduRover V4.1.0-dev (0536be1c)

Mission Planner:    two COM ports are available:
- ArduPilot MAVLink (COM7)
- ArduPilot SLCAN (COM8)

A. USB connection only, no battery (no GPS)

Data > Messages: no script output (SCR_DEBIG_LVL = 3)

3/19/2021 4:42:34 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:42:33 PM : EKF3 IMU1 initialised
3/19/2021 4:42:33 PM : EKF3 IMU1 forced reset
3/19/2021 4:42:26 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:42:24 PM : EKF3 IMU1 initialised
3/19/2021 4:42:24 PM : EKF3 IMU1 forced reset
3/19/2021 4:42:17 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:42:16 PM : EKF3 IMU1 initialised
3/19/2021 4:42:16 PM : EKF3 IMU1 forced reset
3/19/2021 4:42:09 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:42:07 PM : EKF3 IMU1 initialised
3/19/2021 4:42:07 PM : EKF3 IMU1 forced reset
3/19/2021 4:42:02 PM : IMU0: fast sampling enabled 8.0kHz/2.0kHz
3/19/2021 4:42:02 PM : RCOut: PWM:1-13
3/19/2021 4:42:02 PM : MatekH743 003D0026 30305107 33303739
3/19/2021 4:42:02 PM : ChibiOS: a14aa6b0
3/19/2021 4:42:02 PM : ArduRover V4.1.0-dev (0536be1c)
3/19/2021 4:42:01 PM : IMU0: fast sampling enabled 8.0kHz/2.0kHz
3/19/2021 4:42:01 PM : RCOut: PWM:1-13
3/19/2021 4:42:01 PM : MatekH743 003D0026 30305107 33303739
3/19/2021 4:42:01 PM : ChibiOS: a14aa6b0
3/19/2021 4:42:01 PM : ArduRover V4.1.0-dev (0536be1c)
3/19/2021 4:42:01 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:41:59 PM : EKF3 IMU1 initialised
3/19/2021 4:41:59 PM : EKF3 IMU1 forced reset
B. USB and 14.4V NiMH battery (no GPS)

Data > Messages: no script output (SCR_DEBIG_LVL = 3)

3/19/2021 4:44:43 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:44:41 PM : EKF3 IMU1 initialised
3/19/2021 4:44:41 PM : EKF3 IMU1 forced reset
3/19/2021 4:44:34 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:44:33 PM : EKF3 IMU1 initialised
3/19/2021 4:44:33 PM : EKF3 IMU1 forced reset
3/19/2021 4:44:26 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:44:24 PM : EKF3 IMU1 initialised
3/19/2021 4:44:24 PM : EKF3 IMU1 forced reset
3/19/2021 4:44:17 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:44:16 PM : EKF3 IMU1 initialised
3/19/2021 4:44:16 PM : EKF3 IMU1 forced reset
3/19/2021 4:44:09 PM : EKF3 IMU1 tilt alignment complete
3/19/2021 4:44:07 PM : EKF3 IMU1 initialised
3/19/2021 4:44:07 PM : EKF3 IMU1 forced reset
3/19/2021 4:44:04 PM : IMU0: fast sampling enabled 8.0kHz/2.0kHz
3/19/2021 4:44:04 PM : RCOut: PWM:1-13
3/19/2021 4:44:04 PM : MatekH743 003D0026 30305107 33303739
3/19/2021 4:44:04 PM : ChibiOS: a14aa6b0
3/19/2021 4:44:04 PM : ArduRover V4.1.0-dev (0536be1c)
3/19/2021 4:44:04 PM : IMU0: fast sampling enabled 8.0kHz/2.0kHz
3/19/2021 4:44:04 PM : RCOut: PWM:1-13
3/19/2021 4:44:04 PM : MatekH743 003D0026 30305107 33303739
3/19/2021 4:44:04 PM : ChibiOS: a14aa6b0
3/19/2021 4:44:04 PM : ArduRover V4.1.0-dev (0536be1c)
3/19/2021 4:44:01 PM : EKF3 IMU0 tilt alignment complete
3/19/2021 4:44:01 PM : EKF3 IMU1 tilt alignment complete
I'm seeing the same issue with params as well - they are not persisted on reboot.

==
This is what #general advised:
amilcar — Today at 9:44 PM
The main reason for this release is the parameter backup/restore. This has been an ongoing issue for quite some time. The symptom is an unexpected full parameter reset on boards using FRAM (also known as RAMTRON) for storage, particularly those using F7 or H7 based MCUs. The issue has most commonly been seen on the Hex CubeOrange board, but has been seen on boards from other vendors as well.

The issue has been frustratingly difficult to reproduce. We did narrow down one cause last year which was a floating CS pin in the bootloader triggering corruption before ArduPilot starts. An updated bootloader reduced the occurrance of the issue a lot and we thought it was solved. Since then the problem has still happened a few times on boards that have had a bootloader update.

This release avoids the issue by keeping a complete second copy of the parameters in the 2nd half of 32k FRAM devices. On boot we check the integrity of the primary parameter storage area via a simple signature check and if it has been corrupted then we restore from the backup area. We also raise an internal error named" params_restored". You will then need to reboot to clear the error, but your parameters will have been automatically recovered, avoiding the need to reload parameters and re-calibrate.

If you get this internal error then it would be appreciated if you could send us the contents of your APM/STRG_BAK directory on the microSD card so we can analyse the corruption that happened and ensure that the fix covers all real cases.
==

**Version**
ArduRover V4.1.0-dev (6c67e191)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[x] Rover
[  ] Submarine

**Airframe type**
N/A (not selected yet) but intended for a boat

**Hardware type**
Mateksys H743-WING

**Logs**
[APM.zip](https://github.com/ArduPilot/ardupilot/files/6174949/APM.zip)

xcl010, 2021-06-06:
> ## Bug report
> **Issue details**
> Firmware does not retain any parameter setting on https://firmware.ardupilot.org/Rover/latest/MatekH743/ardurover_with_bl.hex for this flight controller. This issue does not occur on the latest arducopter or ArduRover V4.1.0-dev ([37cf09a](https://github.com/ArduPilot/ardupilot/commit/37cf09a8f9665487c755df4b045723860c7e3125)) from 2020-10-18.
> I tested this on 19 March 2021
> Host System: Windows 10
> No peripherals attached, SD card only
> I noticed that two com ports were available, which was unusual as only one should appear.
> 
> I requested that another person try this on their same flight controller and they also had the same results:
> Date: 19 March 2021
> Host system: macOS 10.15.7
> Flight controller: Mateksys H743-WING
> 
> Test 1: known good
> 
> No peripherals attached, SD card only
> 
> Firmware: ArduRover V4.1.0-dev ([37cf09a](https://github.com/ArduPilot/ardupilot/commit/37cf09a8f9665487c755df4b045723860c7e3125))
> Flash: Install with ST32CubeProgrammer after full chip erase
> SD card: SanDisk Ultra microSDHC UHS-I Card, 16 GB, 98 MB/s
> 
> GSC: Mission Planner (running on Windows 10 VMware VM)
> 
> Config > Full Parameter List > SCR_ENABLE = 1 (then reload params and restart)
> Config > MAVFtp > APM > scripts > hello_world.lua
> 
> A. USB connection only, no battery
> 
> Data > Messages: hello world script running correctly
> 
> B. USB and 14.4V NiMH battery
> 
> Data > Messages: hello world script running correctly
> Test 2: install newer version of ArduPilot
> 
> Firmware: ArduRover V4.1.0-dev ([0536be1](https://github.com/ArduPilot/ardupilot/commit/0536be1c889fda6656496419c2b7c2b56efb5ae7))
> 
> Mission Planner: two COM ports are available:
> 
> * ArduPilot MAVLink (COM7)
> * ArduPilot SLCAN (COM8)
> 
> A. USB connection only, no battery (no GPS)
> 
> Data > Messages: no script output (SCR_DEBIG_LVL = 3)
> 
> 3/19/2021 4:42:34 PM : EKF3 IMU1 tilt alignment complete
> 3/19/2021 4:42:33 PM : EKF3 IMU1 initialised
> 3/19/2021 4:42:33 PM : EKF3 IMU1 forced reset
> 3/19/2021 4:42:26 PM : EKF3 IMU1 tilt alignment complete
> 3/19/2021 4:42:24 PM : EKF3 IMU1 initialised
> 3/19/2021 4:42:24 PM : EKF3 IMU1 forced reset
> 3/19/2021 4:42:17 PM : EKF3 IMU1 tilt alignment complete
> 3/19/2021 4:42:16 PM : EKF3 IMU1 initialised
> 3/19/2021 4:42:16 PM : EKF3 IMU1 forced reset
> 3/19/2021 4:42:09 PM : EKF3 IMU1 tilt alignment complete
> 3/19/2021 4:42:07 PM : EKF3 IMU1 initialised
> 3/19/2021 4:42:07 PM : EKF3 IMU1 forced reset
> 3/19/2021 4:42:02 PM : IMU0: fast sampling enabled 8.0kHz/2.0kHz
> 3/19/2021 4:42:02 PM : RCOut: PWM:1-13
> 3/19/2021 4:42:02 PM : MatekH743 003D0026 30305107 33303739
> 3/19/2021 4:42:02 PM : ChibiOS: a14aa6b0
> 3/19/2021 4:42:02 PM : ArduRover V4.1.0-dev ([0536be1](https://github.com/ArduPilot/ardupilot/commit/0536be1c889fda6656496419c2b7c2b56efb5ae7))
> 3/19/2021 4:42:01 PM : IMU0: fast sampling enabled 8.0kHz/2.0kHz
> 3/19/2021 4:42:01 PM : RCOut: PWM:1-13
> 3/19/2021 4:42:01 PM : MatekH743 003D0026 30305107 33303739
> 3/19/2021 4:42:01 PM : ChibiOS: a14aa6b0
> 3/19/2021 4:42:01 PM : ArduRover V4.1.0-dev ([0536be1](https://github.com/ArduPilot/ardupilot/commit/0536be1c889fda6656496419c2b7c2b56efb5ae7))
> 3/19/2021 4:42:01 PM : EKF3 IMU1 tilt alignment complete
> 3/19/2021 4:41:59 PM : EKF3 IMU1 initialised
[... 40 lines cut ...]
> 
> This release avoids the issue by keeping a complete second copy of the parameters in the 2nd half of 32k FRAM devices. On boot we check the integrity of the primary parameter storage area via a simple signature check and if it has been corrupted then we restore from the backup area. We also raise an internal error named" params_restored". You will then need to reboot to clear the error, but your parameters will have been automatically recovered, avoiding the need to reload parameters and re-calibrate.
> 
> # If you get this internal error then it would be appreciated if you could send us the contents of your APM/STRG_BAK directory on the microSD card so we can analyse the corruption that happened and ensure that the fix covers all real cases.
> **Version**
> ArduRover V4.1.0-dev ([6c67e19](https://github.com/ArduPilot/ardupilot/commit/6c67e191d4a03e1da289f531f54915c59b530f51))
> 
> **Platform**
> [ ] All
> [ ] AntennaTracker
> [ ] Copter
> [ ] Plane
> [x] Rover
> [ ] Submarine
> 
> **Airframe type**
> N/A (not selected yet) but intended for a boat
> 
> **Hardware type**
> Mateksys H743-WING
> 
> **Logs**
> [APM.zip](https://github.com/ArduPilot/ardupilot/files/6174949/APM.zip)

https://github.com/ArduPilot/ardupilot/issues/15916#issuecomment-855377469

IamPete1, 2022-08-05:
Sorry we never got to this, please re-open if the issue remains.

closed 2022-08-05 by IamPete1