ArduPilot/ardupilot #16961: No EEPROM, No FRAM, How to storage paramters in SD card

Abner0907, 2021-03-22:
Hi, I am very appreciate that anyone can help me. Now I have a new Flight Controller board(similar as pixhawk V5), the hardware  just support a SD card(8GB+/16gb+), without FRAM or EEPROM. How can I storage the parameters in SD card(SDMMC format). Now the changed parameters will drop after boot.

Again, thank you very much anyone can give me some supports, I am new in ardupilot.

karthikdesai, 2021-03-22:
Please post support questions in the forum. There is no way to store parameters in the SD card. Parameters are stored in persistent chip memory. If the parameters are not persisting, update to the latest stable version.

peterbarker, 2021-03-22:
On Mon, 22 Mar 2021, Abner0907 wrote:

> Hi, I am very appreciate that anyone can help me. Now I have a new Flight Controller board(similar as pixhawk V5), the hardware just support a SD
> card(8GB+/16gb+), without FRAM or EEPROM. How can I [...]

Currently ArduPilot doesn't support storing parameters on SD card.

We have had support for storing parameters in a log-structured filesystem 
in some pages set aside in flash - but that work wasn't merged IIRC.

rmackay9, 2021-03-22:
Seems like a developer question (I guess you're trying to modify AP so that it can run on this board?) in which case it might be good to ask in the chibios channel in discord. http://www.ardupilot.org/discord

closed 2021-03-22 by rmackay9

Abner0907, 2021-03-23:
@rmackay9  Thank you for your reply. Now my hardware can support PX4's work. I also want to modify the code to support the ardupilot. My hardware and Firmware software is open source here:
Hardware: https://github.com/CodevDynamics/Opensource_Hardware
AP/FC: https://github.com/CodevDynamics/Codev-autopilot

Abner0907, 2021-03-23:
@rmackay9  I am very appreciated that you can give me some advice. Now I am not familarity with the Ardupilot and learning the code. Thank you very much.

rmackay9, 2021-03-23:
@Abner0907, we've got a slightly out-of-date wiki page re porting to new flight controller's here: https://ardupilot.org/dev/docs/porting.html

I'd like to help but it's actually far from my area of expertise.  I think @bugobliterator and @tridge are the real experts but there are others like @andyp1per and @Hwurzburg who have done more of this than I have.

Abner0907, 2021-03-23:
@rmackay9 Thank you very much. I will learn the code  firstly and try to let the Ardupilot support our hardware. If success, I will let the code open.  I will ask for help from you and the developers you mentioned above if I have some question which i can not solve or fix.