ArduPilot/ardupilot #16962: pwm noise when rebooting on CUAV X7 with Hardware safety off.

yatink-ci, 2021-03-22:
## Bug report

**issue**
The MultiCopter Motors spin up when rebooting(using GCS - MP and QGC) on a CUAV X7 (and probably on any FC without IO), 

the firmware doesn't wait for the safety to get on and resets just as soon as the command is issued (Probably watchdog reset in delay in function GCS_MAVLINK::handle_preflight_reboot) on copter 4.0.4

**Version**
Copter 4.0.4

**Platform**
[  ] All
[  ] AntennaTracker
[☺] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Hexacopter (x)

**Hardware type**
CUAV X7 with CAN PDB and neo V3 pro

**Logs**

[381 01-01-1980 5.30.00 AM.zip](https://github.com/ArduPilot/ardupilot/files/6181931/381.01-01-1980.5.30.00.AM.zip)

mlebret, 2021-03-23:
Hello,

I have quite the same configuration. CUAV X7 win CAN PDB and Octo setup. Hardware safety switch is disabled (Gimbal control) and ESC 1 to 8 out is set to Oneshot 125. RC3 in  and RC out signal stay clean after reboot (via Mission planner Control F reboot button). I did an arming test to confirm data.

[https://1drv.ms/u/s!AlTaMbgj9DGHgYNwLN00yy29tBnlZw?e=5bsveb](url)

yatink-ci, 2021-03-24:
My ESC's are normal PPM (PWM) Type.

yatink-ci, 2021-03-31:
@rmackay9, Can you please look into it. Thanks.