ArduPilot/ardupilot #16967: position instability after compass loss

obicons, 2021-03-22:
## Bug report

When flying with a quadcopter, I lose compasses during a land sequence. The vehicle's position drifts when this occurs. I expect the vehicle to continue straight down, even without compasses.
[image]

**Version**
HEAD

**Platform**
[  ] All
[  ] AntennaTracker
[*] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
SITL, Iris

**Logs**
https://drive.google.com/file/d/1lZ9wQLzu9hw7geXPFVajBNwB0-OtQ1LG/view?usp=sharing

rmackay9, 2021-03-22:
During a landing it will normally attempt to hold position (using GPS, etc) so the vehicle needs to know its heading.  So if the heading become inaccurate for some reason the vehicle will move around.

Rajeta-Sarkar, 2021-03-28:
Hi ! I'm Rajeta Sarkar and I'm interested in working in this issue. Can you tell me the pre requisites i.e the language and information required to solve this issue. Also where should I start from?