ArduPilot/ardupilot #16981: Automatic raising and lowering of attached equipment

Supermagnum, 2021-03-23:
## Feature request
Automatic raising and lowering equipment attached to front or back.
In the case of a agricultural rover, that is plowing it is necessary to raise the plow at each end of the plow furrow, and turn the rover 180 degrees to start the next set. Then the plow needs to be lowered again.
Some plows also needs to be reversed on each turn. 

A nice YouTube clip that demonstrates this:
https://youtu.be/70SjpOf_h0I

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Additional context**
A plow or other equipment can in some circumstances “bog down” or get filled up with earth the tractor will then start to spin or loose forward momentum and eventually either stall or get stuck. A root or a stone, or a spot of wet ground can cause this.
The only way to clear the problem is to raise the equipment so the “clog” clears.

rmackay9, 2021-03-23:
This could be done to some extent using a DO_SET_SERVO or perhaps DO_GRIPPER mission command placed immediately after the waypoint command.  It might be necessary to add a delay into the waypoint command (first field) or add a NAV_DELAY command after the waypoint to stop the vehicle from moving for a moment in order to give time for the plow to be raised.

Supermagnum, 2021-03-24:
> This could be done to some extent using a DO_SET_SERVO or perhaps DO_GRIPPER mission command placed immediately after the waypoint command. It might be necessary to add a delay into the waypoint [...]

Yes, a linear actuator will solve this if it is connected to the existing operating lever. It can also easily be unhooked for manual operation. No need to alter the hydraulic system.

closed 2021-09-12 by IamPete1