ArduPilot/ardupilot #16982: PTO control

Supermagnum, 2021-03-24:
## Feature request

A power take-off or power takeoff (PTO) is any of several methods for taking power from a power source, such as a running engine, and transmitting it to an application such as an attached implement or separate machine. 

Tractors have these at the rear and sometimes also in front. These are used to power pumps, electric power generator and a multitude of other things.

Tractor PTO's often has selectable rotation speeds,  540 revolutions per minute (rpm) or 1000 rpm depending on the engine rpm.
The PTO off/on is on older tractors controlled by a lever, and in newer model push button. 
The rpm gauge in the operators cabin has these marked, so the correct throttle setting can be applied.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

Tractor PTO types:
The original type (designated as Type 1) calls for operation at 540 revolutions per minute (rpm). A shaft that rotates at 540 rpm has 6 splines on it, and a diameter of 1 3⁄8 inches (35 mm).

Two newer types, supporting higher power applications, operate at 1000 rpm and differ in shaft size. The larger shaft has 20 splines (diameter 1 3⁄4 in or 44 mm) (designated as Type 3), while the smaller has 21 splines (diameter 1 3⁄8 in or 35 mm) (designated as Type 2). Farmers typically differentiate these two types by calling them "large 1000" or "small 1000" as compared to the Type 1 which is commonly referred to as the "540".

Due to ever-increasing horsepower requirements from farm implements, and higher horsepower engines being installed in farm tractors, a still larger type (designated as Type 4) has been added to ISO 500. It operates at a higher rotational speed of 1300 rpm in order to allow for power transfer at reduced levels of torque. The shaft has 22 splines with a major diameter of 57.5 millimeters (mm). It is meant to handle PTO powers up to 450 kilowatts (kW), or roughly 600 horsepower (hp).

All four types rotate counterclockwise when viewed from inside the tractor's cab; when standing behind the tractor, the shaft turns clockwise.

Supermagnum, 2021-03-26:
> Is there a feature request in here somewhere? As far as I can tell this is just a copy and paste from the PTO Wikipedia page?
> 
> What would you like ArduRover to actually do?

Turn the PTO off or on with mission waypoints or manually.

IamPete1, 2021-03-26:
We can already do that with the relay params, see https://ardupilot.org/rover/docs/common-relay.html

closed 2021-03-26 by IamPete1