ArduPilot/ardupilot #17001: Plane: Throttle nudge issues

SlipstreamFPV, 2021-03-25:
Throttle nudge very often causes issues with the throttle control in auto modes. With TECS this can also effect the pitch loop. 

At a guess this occurs when the throttle signal from the radio is not 100% stable and the controller gets into an oscillation by constantly trying to correct up and down. 

Often this is fixed by disabling throttle nudge however then the airspeed is not controllable. 

Possible suggestions to fix this:

- Filter the RC throttle signal before allowing it to demand changes to airspeed, or,

- Implement a deadzone around the throttle position in auto modes. Don’t adjust the target airspeed unless there is a change of a certain percentage, or,

- step the throttle nudge to only allow 5% or even 10% adjustments at a time to airspeed not .1% like currently. 

Thanks for your consideration. I realise these solutions are not perfect, just ideas.

IamPete1, 2021-03-26:
Do you have a log of this? I would be surprised if it could not be resolved with some TECS tuning.

SlipstreamFPV, 2021-03-26:
https://drive.google.com/file/d/1e8HlJqqfjyZE5X5Epf-SakEKP-9FFEG9/view?usp=sharing

It happens in the first part of the flight while in cruise mode. But then doesn’t happen other times.

IamPete1, 2021-03-26:
Your RC in is unusually noisy, might be worth checking that.
[image]

You can see the resulting noise in desired speed and throttle
[image]

A fix would to reduce `THR_SLEWRATE`, but of course that would effect all modes.

SlipstreamFPV, 2021-03-26:
I do agree relating to the noise, this is a taranis x9D plus with Hall effect gimbals, but the Opentx ADC filter on the sticks is turned off because other flight control software works more reliably that way and it’s a system wide setting. Switching it back reduces the noise down to +/- 1 instead of +/- 2.5. I still have the problem either way. 

The issue I have is that no one would ever command a 2,3,4 or 5 us change. Just bumping the sticks is over 10us. So why does the flight control software react to such tiny (much less than 1%) changes? Should it not have some kind of averaging filter on the RC input even if it adds a tiny bit of latency?

WickedShell, 2021-03-27:
Throttle input on plane is an integer from 0-100, a small timing change on the input can step you up or down 1% of the demand, which introduces the noise into the loop. It's an already open issue #2716 so I'm going to close this as a duplicate for now.

closed 2021-03-27 by WickedShell