ArduPilot/ardupilot #17007: Rover: ZigZag mode

Koenstruction, 2021-03-25:
## Feature request

For agricultural rovers, it would be very nice to have the Copter's ZigZag mode work on the Rover platform.

One time setup (per field):
record position 'A' and 'B' only once on the field. The software saves the compass angle and distance between those locations.

Using Zigzag mode:
After manually driving the rover into an agriculture ‘row/line’, we enable the zigzag mode and it should save the actual GPS-position as location A and start to drive to the virtual waypoint B (calculated from location 'A' with the saved distance and compass angle between A & B. The navigation should stay as close to the line between the two points as possible (cross-tracking algorithm).

Extra:
We could drive into the row/line from 2 directions, so it has to calculate if it's heading to the same direction as the saved compass angle, or if it's closer to the opposite compass angle (saved compass angle - 180deg) and use this to calculate waypoint B.
After driving to the end of the row in zigzag mode, it should stop driving and we should be able to turn and drive the rover to the next row manually with the remote and start the ZigZag mode again.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

I have attached a PowerPoint about the desired functional software steps.
[Software ontwerp - Driving mode .pptx](https://github.com/ArduPilot/ardupilot/files/6203259/Software.ontwerp.-.Driving.mode.pptx)

More information from the original Copter ZigZag mode:
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_zigzag.cpp
https://ardupilot.org/copter/docs/zigzag-mode.html
[image]

rmackay9, 2021-03-25:
Seems like a good idea to me!

IamPete1, 2021-03-25:
My vote would be for scripting rather than a proper mode.

muramura, 2021-03-28:
I think I need to verify that the LUA script will work as expected with PIXHAWK1.
The ArduPilot WIKI says that CUBE ORANGE and DURANDAL work well.
I think that if LUA works well with PIXHAWK1, the script is fine.

rmackay9, 2021-03-28:
I'd be happy with it being implemented as a flight mode..