ArduPilot/ardupilot #17022: AP_Scripting: add support for setting EKF origin and home

rmackay9, 2021-03-26:
It would be nice to have a script that allows setting the EKF origin and AHRS home (two separate things) to make it easier for users of AP's improved Non-GPS support including optical flow, T265, etc.

Similar to this issue https://github.com/ArduPilot/ardupilot/issues/7569

srirajshukla, 2021-03-27:
@rmackay9 can you assign this issue to me?

MUzairS15, 2021-04-04:
Sir, i am new to open source can you help me to proceed with this issue?

MUzairS15, 2021-04-04:
I went thtrough the similar issue mentioned and I think we have to define a new function in aclass which is in new library right?

rmackay9, 2021-04-05:
Hi @MUzairS15,

I think we need to add a new line to the [../libraries/AP_Scripting/generator/description/bindings.desc](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Scripting/generator/description/bindings.desc#L29) file and then write a short lua example script ([others examples can be found here](https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_Scripting/examples)) to demonstrate it works.

MUzairS15, 2021-04-05:
i.e i have to add userdata Parameter method set_EKF and set_home 
and then lua example script
 can you explain more anout lua or else can you direct me to doc

rmackay9, 2021-04-05:
Here are our docs on Lua: https://ardupilot.org/copter/docs/common-lua-scripts.html

The change to the bindings.desc is the important change.  You'll need to add a new line for set_home and set_origin.  Maybe if you compare the line (in bindings.desc) for other existing methods you can figure out what the syntax should be.

MUzairS15, 2021-04-06:
Hey I am still having problem writing script tell me i am right or wrong the function definitons of set home and set origin has to be added as example in script?

MUzairS15, 2021-04-06:
I am able to add lines in bindings.desc

rmackay9, 2021-04-07:
@MUzairS15,

You'll definitely need to write a lua script so that you can test that the set-home and set-origin calls work.  you should be able to do that all within SITL.  I'd recommend testing set-home first because that should always work.  set-origin will only work if the vehicle doesn't have a GPS enabled (you'll need to set GPS_TYPE = 0 and maybe change the EK3_SRC1_POSXY, VELXY params to zero too to allow set_origin to work)

MUzairS15, 2021-04-07:
Trying my best but not able to fiigure it out

MUzairS15, 2021-04-09:
https://github.com/ArduPilot/ardupilot/tree/master/libraries/SITL
in this?

MUzairS15, 2021-04-09:
i apologize for irritating you but i really want to do.
I looked for tests so that i can understand the code(flow) how it works but finding it difficult.
Also virtual bool set_origin(const Location &loc) WARN_IF_UNUSED { return false; } is it all to set origin means it always return false so how is it helpful.
I know i am sounding dumb (but as told you just started so want to learn how the organizations built and develop code)
pls guide me

rmackay9, 2021-04-09:
Hi @MUzairS15,

No problem, the AHRS class is only the parent class but there are child classes that override the method so that it does something useful.

CallanDaniel, 2021-06-11:
@rmackay9 I think I have this working, set_home() and get_origin() are working with example scripts. I just can't seem to get set_origin in SITL when passing the function a location object. 

https://github.com/CallanDaniel/ardupilot/tree/pr-scripting-add-set-ekf-origin-and-home

rmackay9, 2021-06-11:
@CallanDaniel,

I can't open that link for some reason but in any case, it is probably failing because some of the EK3_SRC1_xxx parameters are set to GPS or because GPS_TYPE is non-zero so the EKF origin is being set from the GPS data before you have time to set it from the lua script.

CallanDaniel, 2021-06-11:
@rmackay9

Sorry, links broken as I deleted branch because I got a bit messy with the commits (new to all of this). New branch made and will submit pull request shortly. Setting EK2_GPS_TYPE to 3 did the trick.

hendjoshsr71, 2021-06-30:
@CallanDaniel  Were you going to submit the setting of the EKF origin via scripting again? If you did we could close this issue ;)

CallanDaniel, 2021-06-30:
@hendjoshsr71 sure, i'll get that done asap.. appologies for the delay, I've been a bit caught up at work.

CallanDaniel, 2021-06-30:
@hendjoshsr71 just trying to think of an applicable example usage for set_origin to put in the script...

rmackay9, 2021-06-30:
@CallanDaniel,

I would just keep it simple and have it set the origin to some lat, lon, alt.  It could be a place near CMAC in australia (the regular SITL testing location).

CallanDaniel, 2021-06-30:
on it @rmackay9

CallanDaniel, 2021-06-30:
https://github.com/ArduPilot/ardupilot/pull/17907

hendjoshsr71, 2021-07-08:
Closing. Fix merged here: https://github.com/ArduPilot/ardupilot/pull/17907

closed 2021-07-08 by hendjoshsr71