ArduPilot/ardupilot #17030: Slow EK3 Yaw drift

andyp1per, 2021-03-28:
I'm still seeing slow EK3 yaw drift on 4.1-dev copter (e2a350f12ea289c50074ebb3dfb62ca4f12902c9). It's much better than it was, but still not quite as good as EK2. Here is a replay log. The copter rotated through 90o during the course of the hover. This flight was indoors. The copter has GPS but no compass

[log2.zip](https://github.com/ArduPilot/ardupilot/files/6217841/log2.zip)

@rmackay9 @priseborough FYI

chobitsfan, 2021-06-08:
Hi @andyp1per I also see this issue.  The copter fly indoor, no gps, optical flow and compass
It may related to #13326

rmackay9, 2021-09-03:
@andyp1per how do you feel about re-testing this with the latest beta?

andyp1per, 2021-09-03:
Hi @rmackay9 I can have another go, but was still seeing this last time I flew on master even with all Paul's fixes. It's a little bit inconsistent - sometimes it happens and sometimes it doesn't.