ArduPilot/ardupilot #17045: Second GPS become Main GPS on Pixhawk4mini

bys1123, 2021-03-30:
Due to GPS port is serial3 and UART&I2C port is serial2 on Pixhawk4mini, if connect second GPS on UART&I2C port, this second GPS will become GPS1, and main GPS become GPS2, that makes some users very confused or set wrong parameters for these GPS modules.

rmackay9, 2021-03-30:
For the GPS connected to Pixhawk4mini's UART/I2C port to be used though, SERIAL2_PROTOCOL must have been changed from "Mavlink" to "GPS"?  What do you think should have happened?  the GPS connected to serial2 should have become GPS2?.. so whichever GPS is connected to SERIAL3 is always GPS1?

bys1123, 2021-03-30:
Yes, after set SERIAL2_PROTOCOL to GPS, then this GPS on SERIAL2 will become GPS1, main GPS on GPS Port become GPS2, so that some kind of swapped GPS/GPS2 status and GPS_TYPE/GPS_TYPE2 configuration.

So if lock GPS1 with SERIAL3 on Pixhawk4mini will be great.

IamPete1, 2021-03-30:
I don't think there is any nice way to fix this.  We could change the serial ordering, but then it would break everyones existing setup. 

I guess we could add a extra serial protocol option, so you would have gps 1 and 2 for serial protocols. But that is quite allot of effort for this edge case.

I appreciate that its slightly annoying to have to swap over the configs GPS1 to GPS2 but there should be no change in behavior if its setup as GPS 1 or 2. Its also quite unusual that someone adds a second GPS to a already setup vehicle, this is not a issue if you have two GPS's from the start.

bys1123, 2021-03-30:
@IamPete1 Thanks. Maybe add some comments on Pixhawk4mini about dual GPS could makes it more clear.
Even from the start, 10pin GPS also looks more like main GPS, so lots of people can't find which one is main GPS, which is second one.

rmackay9, 2021-03-30:
I agree with @IamPete1 that this is very difficult to change and it could actually confuse users more (at least on other boards) so I think I'll close this.  Instead let's update the Pixhawk4Mini wiki page. https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_HAL_ChibiOS/hwdef/PH4-mini/README.md

closed 2021-03-30 by rmackay9