ArduPilot/ardupilot #17051: Bootloader: Serial Bootloader does not work on CubeOrange

jmachuca77, 2021-03-30:
## Bug report

**Issue details**
uploader.py fails when trying to load firmware through a real serial port to a CubeOrange. 

Test setup, CubeOrange connected to Jetson Xavier serial port at 921600 baud. Confirmed communication is working to mavproxy prior to try to load firmware
```
./uploader.py --port /dev/ttyTHS0 --baud-flightstack 921600 ardurover.apj 
Loaded firmware for 8c,0, size: 1441796 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
```

Cube seems to reboot, but firmware never starts uploading.

For reference same companion computer connected to a cube black:
```
./uploader.py --port /dev/ttyTHS0 --baud-flightstack 921600 ardurover-black.apj 
Loaded firmware for 9,0, size: 1416804 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Attempting reboot on /dev/ttyTHS0 with baudrate=921600...
If the board does not respond, unplug and re-plug the USB connector.
Found board 9,0 bootloader rev 5 on /dev/ttyTHS0
Bootloader Protocol: 5
OTP:
  type: ÿÿÿÿ
  idtype: =FF
  vid: ffffffff
  pid: ffffffff
  coa: //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////8=
  sn: 003f003f3435510738393730
ChipDes:
  family: STM32F42x
  revision: 3
Chip:
  20016419 STM32F42x_43x rev3 (no 1M flaw)
Info:
  flash size: 2080768
  board_type: 9 (fmuv3)
  board_rev: 0
Identification complete

Erase  : [====================] 100.0% (timeout: 7 seconds) 
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.

```

**Version**
Master

**Platform**
[ x ] All

**Hardware type**
CubeOrange

magicrub, 2021-03-31:
Is it the same bootloader on those tests? I suspect they're running at different baudrates. Compile and update the bootloader for both hw and report back

jmachuca77, 2021-03-31:
Both bootloaders were updated to the latest prior to the test. Using mavproxy's flashbootloader command so they should both have the latest one. As far as I understand the bootloader is included with the firmware? And the command just writes it to flash?

jmachuca77, 2021-04-29:
Really need to get this working.

magicrub, 2021-04-29:
Uhm.. I think this was fixed a long time ago. Have you updated your bootloader with master?

jmachuca77, 2021-04-29:
Yes, I am using master. Its only a problem on CubeOrange.

jmachuca77, 2021-04-29:
I also reproduced the issue with an FTDI cable connected directly to my computer with the same results as above, I can hear the cube rebooting, but it just stays with the same output as above... using a cube black with the ftdi cable also works correctly as stated above.

tridge, 2021-04-29:
you didn't say which serial port you are using on the cubeorange. Only two ports are enabled in CubeOrange bootloader:
 - USB
 - serial5 (the 3 pin connector)
you can edit SERIAL_ORDER to add more if you want them

jmachuca77, 2021-04-30:
Thanks Tridge I am using telem1 (serial1) and telem2 (serial2), basically its the same port we use to communicate from the companion computer to the cube. Is there a reason why its different from CubeBlack?

jmachuca77, 2021-04-30:
I think I figured it out... adding this makes it work.... @proficnc  any reason this is not enabled in CubeOrange but it is in black? any reason not to enable it and if not is this a change you can do to your production line?

```
# order of UARTs (and USB)
SERIAL_ORDER OTG1 USART2 USART3 UART7

# Another USART, this one for telem1. This one has RTS and CTS lines.
# USART2 serial2 telem1
PD3 USART2_CTS USART2
PD4 USART2_RTS USART2
PD5 USART2_TX USART2
PD6 USART2_RX USART2

# The telem2 USART, also with RTS/CTS available.
# USART3 serial3 telem2
PD8 USART3_TX USART3
PD9 USART3_RX USART3
PD11 USART3_CTS USART3
PD12 USART3_RTS USART3
```

proficnc, 2021-04-30:
All good to add to me, double check with @bugobliterator.

PR #17320 (2021-11-10): HAL_ChibiOS: CubeOrange Enable Serial2 and Serial3 in bootloader hwdef
In the CubeOrange bootloader serial ports 2 and 3 (telem1 and 2) are disabled so they cannot be used to upload firmware. This ports are enabled on the CubeBlack bootloader. This PR enables them for the CubeOrange.
Fixes #17051
closed 2021-11-10 by tridge