ArduPilot/ardupilot #17078: [Copter] Virtual cable cam

robustini, 2021-04-01:
I have often had to take video footage along a route previously created in the Mission Planner flight plan, often between just two waypoints as if it were a virtual cable cam. 
Since these shots had to be repeated I created two or more waypoints with the mission in loop via the DO_JUMP.
Definitely convenient, but you must always interact on Mission Planner, even if i want to change the path.
Instead, it would be useful to be able to do it completely from the remote control.
We already have the ability to add waypoints in flight with a switch, what is missing is the ability to delete the stored mission to then add new waypoints,  and the ability to put it in loop.
So I ask for this functionality to be implemented.
This is the operating logic:

- I takeoff and save two or more waypoints with the switch via CH_OPT
- I switch to AUTO and the copter starts in the mission but by looping on the saved waypoints, to do this it would be enough for AUTO to read the high state condition of the channel used to memorize the waypoints, if active the mission goes in loop
- through a new function in CH_OPT I can delete the recorded mission to be able to store a new one during the flight, or something in any case that allows it

peterbarker, 2021-04-02:
On Thu, 1 Apr 2021, Marco Robustini wrote:

> We already have the ability to add waypoints in flight with a switch, what is missing is the ability to delete the stored mission to then add new waypoints, and the ability to put it in
> loop.

You can definitely clear the mission - it's rc channel option #58, 
"CLEAR_WP"

No ability to add a loop in - and its tricky because there's the question 
of which point to loop back to.

Waiting for the chorus of "scripting!" :-)

robustini, 2021-04-02:
Thanks @peterbarker, option 58 had escaped me, this is because we are starting to have so many! :-)
Regarding the loop: from the first ID in the list, because I don't see anything dangerous or dramatic, the loop will do anyway.