ArduPilot/ardupilot #17090: Return from _ahrs.get_quat_body_to_ned(attitude_vehicle_quat) seems wrong

lthall, 2021-04-03:
I have just finished rewriting the attitude controller quaternion correction in an attempt to clean up the math and make it easier to read. In doing so I found one of the reasons I struggled to get everything working correctly.

I used _ahrs.get_quat_body_to_ned(quat_body_to_ned) in the original version but when I attempted to clean up the code I found that it did not give me the same results as the attitude of the aircraft. I was also getting some strange recovery paths in my rewrite. After adding _ahrs.get_quaternion(attitude_vehicle_quat) all my code started working correctly.

I added logging to compare:
attitude_vehicle_quat
to:
quat_body_to_ned.inverse()

I would think these should be the same but they are obviously not. I am pretty sure I just don't know what I am looking at here rather than having a bug but any explanation would be great!

```
 // Retrieve quaternion vehicle attitude
    Quaternion attitude_vehicle_quat;
    if (!_ahrs.get_quaternion(attitude_vehicle_quat)){
        return;
    }

    Quaternion quat_body_to_ned;
    _ahrs.get_quat_body_to_ned(quat_body_to_ned);
    AP::logger().Write("QAT",
        "TimeUS,r,p,y,nir,nip,niy",
        "sddhddh",
        "F000000",
        "Qffffff",
          AP_HAL::micros64(),
          (double)degrees(attitude_vehicle_quat.get_euler_roll()),
          (double)degrees(attitude_vehicle_quat.get_euler_pitch()),
          (double)degrees(wrap_2PI(attitude_vehicle_quat.get_euler_yaw())),
          (double)degrees(quat_body_to_ned.inverse().get_euler_roll()),
          (double)degrees(quat_body_to_ned.inverse().get_euler_pitch()),
          (double)degrees(wrap_2PI(quat_body_to_ned.inverse().get_euler_yaw())));
```

attitude_vehicle_quat matches what we expect to see:
[image]

quat_body_to_ned.inverse() is not only inverted but also seems to follow a different path suggesting it is actually something else completly.
[image]

esaldiran, 2021-04-03:
Hi @lthall,

get_quat_body_to_ned, get_quaternion, and ATT logs all uses the following variable to populate vehicle orientation and all of them define orientation from body to inertial frame. These functions get their data from the respective EKF instance. 

[image]
Comment above getQuaternion is contradicting with getRotationBodyToNED function above it. I think it also needs changing. 

Both get_quat_body_to_ned and ATT logs take into account AHRS_TRIM parameters, but getQuaternion does not. From your first image, it can be seen your trim parameters are zero.

For the second image, quat_body_to_ned.inverse() is the inverse of orientation presented by ATT logs, it is normal to get different result. 

Since all of these attitude representation methods uses outputDataNew.quat variable in EKF, they should all correspond to the same thing, unless AHRS_TRIM are set to non zero values.

esaldiran, 2021-04-03:
@lthall I compare get_quat_body_to_ned, get_quaternion and ATT logs. They overlap with each other. ATT logs are sampled with at time, so I added ATT field to QAT logging.

Here is the logging function I used. 

    Quaternion attitude_vehicle_quat;
    if (!_ahrs.get_quaternion(attitude_vehicle_quat)){
        return;
    }

    Quaternion quat_body_to_ned;
    _ahrs.get_quat_body_to_ned(quat_body_to_ned);
    AP::logger().Write("QAT",
        "TimeUS,r,p,y,nir,nip,niy,ar,ap,ay",
        "sddhddhddh",
        "F000000000",
        "Qfffffffff",
          AP_HAL::micros64(),
          (double)degrees(attitude_vehicle_quat.get_euler_roll()),
          (double)degrees(attitude_vehicle_quat.get_euler_pitch()),
          (double)degrees(wrap_2PI(attitude_vehicle_quat.get_euler_yaw())),
          (double)degrees(quat_body_to_ned.get_euler_roll()),
          (double)degrees(quat_body_to_ned.get_euler_pitch()),
          (double)degrees(wrap_2PI(quat_body_to_ned.get_euler_yaw())),
          (double)degrees(_ahrs.roll),
          (double)degrees(_ahrs.pitch),
          (double)degrees(wrap_2PI(_ahrs.yaw)));

They all overlap with each other. 
[image]

If we looked at the difference between get_quaternion and get_quat_body_to_ned, and between get_quat_body_to_ned and ATT logs, we find there is none. 
[image]

lthall, 2021-04-03:
The name get_quat_body_to_ned suggests that this is the rotation to the inertial frame from the body frame. But this is the rotation from the inertial frame to the body frame. This also appears to be calculated based on the rotation matrix that is also labeled as body frame to NED.

It looked to me like the rotation matrix was being used to rotate sensors back to the inertial frame so what am I missing here. Maybe I just haven't been getting enough sleep.

esaldiran, 2021-04-04:
Unlike your logging code, I did not take the inverse of get_quat_body_to_ned while calculating the euler angle. Ex:
Yours,
 (double)degrees(quat_body_to_ned.inverse().get_euler_roll())
Mine,
 (double)degrees(quat_body_to_ned.get_euler_roll())

This is why everything overlaps and all of the rotation representations be it a rotation matrix or a quaternion are defined from body frame to inertial frame.

lthall, 2021-04-04:
Yes, I just looked at this and got the same result. 
The name seems wrong and it worries me about the rotation matrix it is based on. I am concerned that the rotation matrix is being used in the wrong way now.
Again, maybe I am getting the math wrong.

esaldiran, 2021-04-04:
I am sorry, but I don't understand exactly which rotation matrix you are referring to. All of the functions asking for vehicle orientation derived from the same source, thus they are a different ways of representing the same thing.

lthall, 2021-04-04:
The names imply that this is the rotation from the body frame back to the NED frame:
_ahrs.get_quat_body_to_ned(quat_body_to_ned); 

Yet we have shown that they are actually the rotation from the NED frame to the body frame.

If you follow the code back to see where this quaternion is calculated it comes from a rotation matrix that is also labeled as body_to_ned. When I looked at where this rotation matrix was used it I found it appears to be used to rotate some sensor data from the body frame back to the NED frame.

My concern is that this rotation matrix may not be doing what it's name suggests and this sensor data is being rotated away from the NED instead of to it.

esaldiran, 2021-04-04:
I believe there is a misunderstanding here. _ahrs.get_quat_body_to_ned(quat_body_to_ned) is rotation from body to NED frame. 

As I have shown in my first answer, the Euler angles generated from _ahrs.get_quat_body_to_ned and get_quaternion match with each other. Since they are not different from one another and they are derived from the same source, we can't say one is rotation from NED to body but the other is from the body to NED. 

If you are referring to get_rotation_body_to_ned function, yes it is used at multiple library. When I took a glance at these libraries, I didn't encounter miss use of get_rotation_body_to_ned function. There was some example in TECS library that I didn't understand, but they would require further inspection for me to say anything.

lthall, 2021-04-05:
Ok, I either have not slept enough or I am stupid......

Let's assume I am missing something really stupid that you will roll your eyes at and wonder how I could ever contribute anything useful to the project.....

> _ahrs.get_quat_body_to_ned(quat_body_to_ned) is rotation from body to NED frame.

I would expect if this was the rotation from the body to the NED frame that if I converted this to roll, pitch, and yaw I would see the opposite to what the ATT logs show. For example, if I was rolled to the right by 45 degrees (+45 degrees), then quat_body_to_ned would produce -45 degrees when converted to roll, pitch and yaw....

When I read through the code we have been discussing I am picturing attitude_vehicle_quat as being FROM NED to the vehicle body frame. Maybe everything is backwards but still works......

lthall, 2021-04-06:
This all makes sense to now.

closed 2021-04-06 by lthall