ArduPilot/ardupilot #17105: EKF has erratic position behaviour on regain of GPS

peterbarker, 2021-04-05:
## Bug report

**Issue details**

https://github.com/ArduPilot/ardupilot/pull/17104 contains an autotest which is currently failing - the EKF forces a second failsafe where it probably should not, instead doing a position-reset or similar.

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad

**Hardware type**
SITL

**Logs**

peterbarker, 2021-04-05:
[ArduCopter-RefindGPS-00000003.TXT](https://github.com/ArduPilot/ardupilot/files/6257180/ArduCopter-RefindGPS-00000003.TXT)

That's actually a `LOG_REPLAY` .BIN file of the failing flight - not an 8.6MB text file :-)

peterbarker, 2021-04-05:
[image]

priseborough, 2021-05-13:
The EKF does start using GPS again based on innovation sequences and solution status.

[image]

[image]

[image]

The solution status value of 107327 maps to the following flags true

        bool attitude           : 1; // 0 - true if attitude estimate is valid
        bool horiz_vel          : 1; // 1 - true if horizontal velocity estimate is valid
        bool vert_vel           : 1; // 2 - true if the vertical velocity estimate is valid
        bool horiz_pos_rel      : 1; // 3 - true if the relative horizontal position estimate is valid
        bool horiz_pos_abs      : 1; // 4 - true if the absolute horizontal position estimate is valid
        bool vert_pos           : 1; // 5 - true if the vertical position estimate is valid
        bool pred_horiz_pos_rel : 1; // 8 - true if filter expects it can produce a good relative horizontal position estimate - used before takeoff
        bool pred_horiz_pos_abs : 1; // 9 - true if filter expects it can produce a good absolute horizontal position estimate - used before takeoff
        bool using_gps          : 1; // 13 - true if we are using GPS position
        bool gps_quality_good   : 1; // 15 - true if we can use GPS for navigation
        bool initalized         : 1; // 16 - true if the EKF has ever been healthy

and the following false:

        bool terrain_alt        : 1; // 6 - true if the terrain height estimate is valid
        bool const_pos_mode     : 1; // 7 - true if we are in const position mode
        bool takeoff_detected   : 1; // 10 - true if optical flow takeoff has been detected
        bool takeoff            : 1; // 11 - true if filter is compensating for baro errors during takeoff
        bool touchdown          : 1; // 12 - true if filter is compensating for baro errors during touchdown
        bool gps_glitching      : 1; // 14 - true if GPS glitching is affecting navigation accuracy

Fault status is zero and the only timeout is for the missing airspeed sensor which doesn't affect copter.

priseborough, 2021-05-13:
There are two short periods after regaining GPS where a different solution status is present:

111423 - touchdown goes true
109375 - takeoff goes true

peterbarker, 2021-05-17:
[image]

peterbarker, 2021-05-18:
This also happens on EKF2.

rmackay9, 2022-02-16:
I think we can close this can't we?  I think PeterB's test was simply trying to use the position estimate before the EKF had recovered.

priseborough, 2022-02-22:
Agree

closed 2022-02-22 by rmackay9

peterbarker, 2022-03-03:
> I think we can close this can't we? I think PeterB's test was simply trying to use the position estimate before the EKF had recovered.

@rmackay9 actually, this issue got fixed sometime in master - the PR showing the bug up in an autotest was committed so it shouldn't reappear.