ArduPilot/ardupilot #17116: Rover: 4.1.0 issues list

rmackay9, 2021-04-07:
This is a list of known issues especially blockers for the 4.1.0 rover release.  The equivalent Copter issue is https://github.com/ArduPilot/ardupilot/issues/16478

- [ ] EKF issues mentioned on the Copter-4.1 issue https://github.com/ArduPilot/ardupilot/issues/16478

yuri-rage, 2021-06-02:
I connected uCenter today and used Tools->Receiver Configuration->GNSS->File to save these four configuration files.

They are labeled accordingly with the GPS_AUTO_CONFIG parameter value in the filenames.  A diff of the moving base (MB) files and Rover files reveals no obvious answers - they are identical.

However, as I was downloading them with the flight controller active and telemetry connected, I had no 'Unhealthy GPS Signal' warnings when GPS_AUTO_CONFIG was enabled.  When I rebooted with it disabled, the 'Unhealthy GPS Signal' warnings were present even after full EKF3 alignment and a good GPS yaw solution.  Without rebooting, I enabled GPS_AUTO_CONFIG, and the message instantaneously disappeared.

[UBX_F9P_Auto=1_MB.txt](https://github.com/ArduPilot/ardupilot/files/6587327/UBX_F9P_Auto.1_MB.txt)
[UBX_F9P_Auto=1_Rover.txt](https://github.com/ArduPilot/ardupilot/files/6587328/UBX_F9P_Auto.1_Rover.txt)
[UBX_F9P_Auto=0_MB.txt](https://github.com/ArduPilot/ardupilot/files/6587329/UBX_F9P_Auto.0_MB.txt)
[UBX_F9P_Auto=0_Rover.txt](https://github.com/ArduPilot/ardupilot/files/6587330/UBX_F9P_Auto.0_Rover.txt)

NickNunno, 2021-09-24:
@rmackay9 has anyone validated whether "Speed controller I-term build-up when motors are saturated" is a legit bug?

geofrancis, 2021-12-14:
the DSHOT reverse range is inverted issue is only on 4.2, it works on 4.1