ArduPilot/ardupilot #17126: NAV_LOITER_TURNS Broken on master

giacomo892, 2021-04-08:
## Bug report

NAV_LOITER_TURNS WP type doesn't work on master with default params.

The copter reaches the WP and waits there forever without starting to circle.

I've tracked down the issue in the fact that the navigation gets "stuck" inside the function:
bool ModeAuto::verify_circle(const AP_Mission::Mission_Command& cmd)

with copter.wp_nav->reached_wp_destination() being false.

working revision: 67f2c79717ff3385bd56ba5c27c0d8c920abf233 before the SCURVES PR

[image]

broken behaviour on master:

[image]

Waypoint file to test with

[way.txt](https://github.com/ArduPilot/ardupilot/files/6279608/way.txt)

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

I think @rmackay9 @lthall and @IamPete1 can have a say on this

shellixyz, 2021-04-08:
I can reproduce the issue

rmackay9, 2021-04-09:
Great, thanks for the report.  I've added this to the 4.1 issues list (https://github.com/ArduPilot/ardupilot/issues/16478) so it won't be forgotten.

giacomo892, 2021-04-10:
Oddly enough it worked in real life but it is broken on SITL at least for me and @shellixyz [image]

rmackay9, 2021-04-11:
@giacomo892,

thanks again for this report and I've got a fix now I think.  maybe you could give it a try when you get a second?
https://github.com/ArduPilot/ardupilot/pull/17142

The problem only happened if lat,lon,alt were specified so perhaps that's why it only happened in SITL.

giacomo892, 2021-04-11:
My pleasure @rmackay9 ! It seems to work fine on SITL now. Will fly test it as soon weather allows. Thanks again for this prompt fix!

rmackay9, 2021-04-11:
fixed in master, thanks for the report and testing!

closed 2021-04-11 by rmackay9