ArduPilot/ardupilot #17157: Anti spin and stability control.

Supermagnum, 2021-04-13:
## Feature request
Anti spin:
The spinning wheel(s) is slowed with less PWM or braking to its motors ESC, diverting more torque to the non-spinning wheels. 

Electronic stability control:
Braking or less PWM is automatically applied to wheels individually, such as the outer front wheel to counter oversteer, or the inner rear wheel to counter understeer. 

The reason for the combination of a request for these features is that they are related to each other.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

mikerob, 2021-04-14:
A very useful first step for this would be single-pin wheel speed encoding via simple pulses (rather than quadrature encoding that needs two pins per wheel). For this functionality you don't need wheel direction, only speed.

In addition to anti-spin, it will also give you ABS.

There's already a level of stability control to counter under- or over-steer via the steering PID yaw rate controller.

geofrancis, 2021-09-14:
I think what Supermagnum is asking about is traction control. vesc speed controllers have traction control when more than one is linked with canbus, they work by setting a maximum RPM difference allowed between the motors before power is reduced to the fastest motor. so for ardurover to do it it would need either wheel speed sensors or a better option would be ESC telemetry to get the rpm of the motors, that way there is no need for external sensors on the motors.

geofrancis, 2021-10-03:
a much simpler way of doing this would be to use a motor controller with telemetry and a script.

haplm, 2021-10-04:
@geofrancis - do you know about a ground esc with RPM telemetry, decent braking, and possibly reverse?