ArduPilot/ardupilot #17167: EK3_DRAG_BCOEF_Y param are reset to the default

tatsuy, 2021-04-13:
## Bug report

**Issue details**
I found a weird issue where EK3_DRAG_BCOEF_Y are reset to the default.

The reproduce procedure:
 1. run STIL
 2. param set EK3_DRAG_BCOEF_Y 50
 3. reboot SITL
 4. param show EK3_DRAG_BCOEF_Y
  displayed "EK3_DRAG_BCOEF_Y 0.000000"

What I confirmed:
 1. delete eeprom.bin
  nothing change
 2. confirmed other EK3_DRAG_* params
  other parameters hold the value after rebooting
 3. changed the index from 3 to 7
  before
`AP_GROUPINFO("DRAG_BCOEF_Y", 3, NavEKF3, _ballisticCoef_y, 0.0f),`
  after
`AP_GROUPINFO("DRAG_BCOEF_Y", 7, NavEKF3, _ballisticCoef_y, 0.0f),`
 DRAG_BCOEF_Y holds the value after rebooting

https://github.com/ArduPilot/ardupilot/blob/c83774a7cd83929b90c772beec268d0fc1bf94d2/libraries/AP_NavEKF3/AP_NavEKF3.cpp#L679

 4. changed the index of DRAG_M_NSE from 4 to 3
  DRAG_M_NSE are reset to the default

https://github.com/ArduPilot/ardupilot/blob/c83774a7cd83929b90c772beec268d0fc1bf94d2/libraries/AP_NavEKF3/AP_NavEKF3.cpp#L688

 5. reproduced this issue with KakuteF7mini

**Version**
latest master (1fe1e532)

**Platform**
[  ] All
[  ] AntennaTracker
[●] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
I confirmed this with quad.

**Hardware type**
SITL (native and WSL Ubuntu 18.04) / KakuteF7mini

rmackay9, 2021-04-14:
I've confirmed this issue in SITL and added it to the Copter-4.1 issues list.

tatsuy, 2021-04-14:
Thank you.
I updated the description.

priseborough, 2021-04-19:
Moving the EK3_DRAG_BCOEF_Y parameter to slot 7 fixes this issue, but the fact that this works indicates a potential bug with the param system.

tridge, 2021-04-20:
> Moving the EK3_DRAG_BCOEF_Y parameter to slot 7 fixes this issue, but the fact that this works indicates a potential bug with the param system.

this is fixed in #17228

closed 2021-04-20 by tridge