ArduPilot/ardupilot #17170: Cube Black - No Light Weight Telemetry output

andypnz, 2021-04-14:
## Bug report

No Light Weight Telemetry output - Black Cube on mini carrier board

**Issue details**

Black Cube on a mini carrier board. Firmware Plane 4.0.9 official.

Trying to use Telemetry2 for Light Weight Telemetry output. So that’s Serial2 - set to 9600 baud (have also tried 1200, 2400), and protocol to 25.

Analyzing the second pin (the pin beside VCC) - which should be Serial2 Tx - and I see nothing.

I have used Light Weight Telemetry on other boards (eg OmniBus F4) no issue.

**Version**
Tried all available firmware for Cube Black

**Platform**
[  ] All
[  ] AntennaTracker  (not tested)
[ Y ] Copter
[ Y ] Plane
[Y  ] Rover
[  ] Submarine  (not tested)

**Airframe type**
fixed wing

**Hardware type**
Cube Black

**Logs**
_Please provide a link to any relevant logs that show the issue_

WickedShell, 2021-04-14:
That protocol isn't available in 4.0, you will need to use master, or wait until 4.1 comes out.

closed 2021-04-14 by WickedShell

andypnz, 2021-04-15:
> That protocol isn't available in 4.0, you will need to use master, or wait until 4.1 comes out.

Thank you. I just loaded 4.1.0 Dev from today, and it is working.

It is a bit of a trick that it was listed in the protocol selection :-)

rmackay9, 2021-04-15:
@andypnz, yes, that is tricky.  It's a long standing issue that we don't have a really good way to separate the stable vs latest parameters.  We will resolve it eventually I hope.