ArduPilot/ardupilot #17174: Windvane library uses average ADC readings with continuous-rotation potentiometers

arikrupnik, 2021-04-14:
## Bug report

Windvane library uses average ADC readings. The library more or less assumes the use of continuous-rotation potentiometers. This combinations creates artificial spikes in reported values.

**Issue details**

Continuous-rotation potentiometers go rapidly from highest to lowest reading as they rotate through the dead zone. Windvane library uses `voltage_average_ratiometric()` to smooth some of ADC noise. Averaging values over the null transition creates artificial spikes. As an example, as the pot transitions from 0.0V (perhaps representing 3 degrees) to 3.3V (perhaps representing 356 degrees), the "average" value would be 1.65V, representing 180 degrees.

_Please describe the problem_

Two graphs below illustrate the problem. Top is an analog voltage output from the pot. Bottom is windvane angle from the same input. The graphs are 180 degrees out of phase in the sense that voltage values overflow from 0V to 3.3V at North while wind direction overflows from 180 degrees to -180 at South. The "average" spikes are clearly visible on the bottom graph at 5, 11 and 15 seconds.

I can see two ways of solving this problem:

1. Use latest instead of averaged values in the Windvane library and filter the ADC noise downstream
2. Amend the averaging code in `AP_HAL_*/AnalogIn.cpp` to reset the averaging sum if the sum is close to one end of the scale and the incoming sample is close to the other.

[image]

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[X] Rover
[  ] Submarine

**Airframe type**
all

**Hardware type**
all

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2021-04-14:
I have added a commit to https://github.com/ArduPilot/ardupilot/pull/17139 that switches to use `voltage_latest()`. Would be nice if we could use the latest ratiometric value, but probably a very marginal improvement for quite a large code change to add that.

arikrupnik, 2021-04-14:
Thank you, @IamPete1. I will test as soon as I can get to the boat.

A part of me feels that the Analog library should handle continuous-rotation pots correctly, although windvane may be the only place where ArduPilot is using them currently.

arikrupnik, 2021-04-15:
Thank you, @IamPete1. This is a huge improvement. There are still artefacts, although they are much smaller.

[image]

IamPete1, 2021-07-18:
I think we fixed this?

closed 2021-07-18 by IamPete1

arikrupnik, 2021-07-18:
Has there been another commit on this since April? Last I checked (see scerrnshot from April 14) there were still noticeable artifacts.

IamPete1, 2021-07-18:
@arikrupnik No more commits, but were now using voltage latest so the artifact must be a genuine voltage reading?

If that is the genuine voltage we cant do anymore in AP than apply the filtering that we already do, provided that the filter is all working as it should....

arikrupnik, 2021-07-18:
How can we explain the visible artifacts then? The pot doesn't arbitrarily ratchet back as it passes though null.

IamPete1, 2021-07-18:
It cant get from full voltage to 0 instantly, in my experience the pin is actually left floating for abit. It would be interesting to see a scope trace, and see how it matches. Maybe a pull down/up would help, we have a dead zone param....