ArduPilot/ardupilot #17190: Rover/Boat: DPTH logging should happen even if not position

rmackay9, 2021-04-15:
The DPTH logging will only happen if the vehicle has a good position estimate.  Even without lat,lon though we should log the depth because it could be useful.  As a minimum it tells the users that the sensor is working and being logged so if there is no position, let's just log the lat,lon as 0.

This should be an easy fix by simply removing the early return.  Search for "DPTH" in ardupilot and you should find where this is happening.

srmainwaring, 2021-04-15:
@rmackay9 agree that it is useful. There's a technique for navigating in fog with only a depth sounder and chart that involves heading to a depth from your last known position then following a contour - not expecting AP to have chart details available but a companion computer might incorporate this into a position estimator.

rmackay9, 2021-05-14:
This is resolved now in master so closing

closed 2021-05-14 by rmackay9