ArduPilot/ardupilot #17195: Intermittent Replay failure in CI

peterbarker, 2021-04-15:
## Bug report

**Issue details**

We've had an intermittent failure in CI for some months,  `check_replay failed`

Yesterday I found I could reliably reproduce the issue.

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine
[X] Replay

**Airframe type**
Copter +

**Hardware type**
SITL

**Steps to Reproduce**

`./Tools/autotest/autotest.py build.Copter test.CopterTests2b --valgrind`

That should fail.

Past that, this should also fail:
```
./waf replay &&
rm logs/* ; "build/sitl/tools/Replay" /mnt/storage/buildlogs/ArduCopter-Replay-00000012.BIN &&
./Tools/Replay/check_replay.py logs/00000001.BIN
```

My current git hash is `ec11bf1146c3ccb39e39f9a86cc3ceb077e32571`

peterbarker, 2021-04-15:
Removing `eeprom-replay.bin` allows the test to pass again.

Compiling with debug stops it happening.

peterbarker, 2021-04-15:
Received this segfault this-morning:

```
pbarker@recursion:~/rc/ardupilot(master)$ ./waf replay &&
> rm logs/* ; "build/sitl/tools/Replay" /mnt/storage/buildlogs/ArduCopter-Replay-00000012.BIN &&
> ./Tools/Replay/check_replay.py logs/00000001.BIN 
Waf: Entering directory `/home/pbarker/rc/ardupilot/build/sitl'
Embedding file locations.txt:Tools/autotest/locations.txt
Embedding file models/Callisto.json:Tools/autotest/models/Callisto.json
[739/743] Compiling Tools/Replay/Replay.cpp
[743/743] Linking build/sitl/tools/Replay
Waf: Leaving directory `/home/pbarker/rc/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/pbarker/rc/ardupilot/build/sitl
Target        Text     Data   BSS     Total  
---------------------------------------------
tools/Replay  1578876  53856  180560  1813292

Build commands will be stored in build/sitl/compile_commands.json
'replay' finished successfully (5.494s)
Suggested EK3_BCOEF_* = 16.288, EK3_MCOEF = 0.209
Starting sketch 'Replay'
Starting SITL input
Starting
ERROR: segmentation fault - aborting
Running: sh Tools/scripts/dumpstack.sh 802305 >dumpstack_Replay.802305.out 2>&1
Stack dumped
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
0x00007fa6226cbdba in __GI___wait4 (pid=802306, stat_loc=stat_loc@entry=0x7ffcadacb7e8, options=options@entry=0, usage=usage@entry=0x0) at ../sysdeps/unix/sysv/linux/wait4.c:27
27	../sysdeps/unix/sysv/linux/wait4.c: No such file or directory.
#0  0x00007fa6226cbdba in __GI___wait4 (pid=802306, stat_loc=stat_loc@entry=0x7ffcadacb7e8, options=options@entry=0, usage=usage@entry=0x0) at ../sysdeps/unix/sysv/linux/wait4.c:27
        resultvar = 18446744073709551104
        sc_ret = <optimised out>
#1  0x00007fa6226cbd7b in __GI___waitpid (pid=<optimised out>, stat_loc=stat_loc@entry=0x7ffcadacb7e8, options=options@entry=0) at waitpid.c:38
No locals.
#2  0x00007fa62263b0e7 in do_system (line=<optimised out>) at ../sysdeps/posix/system.c:172
        __result = <optimised out>
        _buffer = {__routine = 0x7fa62263b290 <cancel_handler>, __arg = 0x7ffcadacb7f0, __canceltype = 0, __prev = 0x0}
        _avail = 1
        cancel_args = {quit = 0x7fa6227d4200 <quit>, intr = 0x7fa6227d42a0 <intr>, pid = 802306}
        status = -1
        ret = 0
        pid = 802306
        sa = {__sigaction_handler = {sa_handler = 0x1, sa_sigaction = 0x1}, sa_mask = {__val = {65536, 0 <repeats 15 times>}}, sa_flags = 0, sa_restorer = 0x0}
        omask = {__val = {1024, 9196350439090552833, 4294967295, 9196350441231745024, 140351517953824, 140351523550224, 140723222264912, 140723222265011, 140723222264880, 140723222265011, 3832619573542697216, 13904296683712223744, 140723222264912, 140351520071872, 94329904227163, 140723222264912}}
        reset = {__val = {6, 0 <repeats 15 times>}}
        spawn_attr = {__flags = 12, __pgrp = 0, __sd = {__val = {6, 0 <repeats 15 times>}}, __ss = {__val = {1024, 9196350439090552833, 4294967295, 9196350441231745024, 140351517953824, 140351523550224, 140723222264912, 140723222265011, 140723222264880, 140723222265011, 3832619573542697216, 13904296683712223744, 140723222264912, 140351520071872, 94329904227163, 140723222264912}}, __sp = {sched_priority = 0}, __policy = 0, __pad = {0 <repeats 16 times>}}
        __cnt = <optimised out>
        __set = <optimised out>
        __cnt = <optimised out>
        __set = <optimised out>
#3  0x000055cae465cfb2 in AP_HAL::dump_stack_trace() ()
No symbol table info available.
#4  0x000055cae465d8dc in _sig_segv(int) ()
No symbol table info available.
#5  <signal handler called>
No locals.
#6  0x000055cae463ce4d in apfs_fprintf ()
No symbol table info available.
#7  0x000055cae45be0e4 in ReplayVehicle::load_parameters() ()
No symbol table info available.
#8  0x000055cae4612889 in AP_Vehicle::setup() ()
No symbol table info available.
#9  0x000055cae45bea7e in Replay::setup() ()
No symbol table info available.
#10 0x000055cae465386d in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
No symbol table info available.
#11 0x000055cae45b57a4 in main ()
[... 27 lines cut ...]
No symbol table info available.
#5  <signal handler called>
No locals.
#6  0x000055cae463ce4d in apfs_fprintf ()
No symbol table info available.
#7  0x000055cae45be0e4 in ReplayVehicle::load_parameters() ()
No symbol table info available.
#8  0x000055cae4612889 in AP_Vehicle::setup() ()
No symbol table info available.
#9  0x000055cae45bea7e in Replay::setup() ()
No symbol table info available.
#10 0x000055cae465386d in HAL_SITL::run(int, char* const*, AP_HAL::HAL::Callbacks*) const ()
No symbol table info available.
#11 0x000055cae45b57a4 in main ()
No symbol table info available.
A debugging session is active.

	Inferior 1 [process 802305] will be detached.

Quit anyway? (y or n) [answered Y; input not from terminal]
[Inferior 1 (process 802305) detached]
Aborted (core dumped)
pbarker@recursion:~/rc/ardupilot(master)$ 

```

IamPete1, 2021-09-12:
@peterbarker Is this still a issue?

peterbarker, 2021-09-14:
On Sun, 12 Sep 2021, Peter Hall wrote:

> @peterbarker Is this still a issue?

I haven't seen any recently.  Which is sad because I can't point to a 
particular commit we made which fxed anything.

Suggest we close this and reopen it if it happens again (much more likely 
if we close this).

closed 2021-09-15 by IamPete1