ArduPilot/ardupilot #17200: Flight Mode change works only once via RC input

wucke13, 2021-04-16:
## Bug report

**Issue details**

When I assign channel 5 as flight mode channel, the flightmode is stuck after the first change via the rc input channel (e.g. after boot, I can switch from the initial mode to another mode __once__). Moving the switch after the first change, won't change the flightmode. According to the GCS the change of the FM rc channel is visible, but the FM does not change anyways. Changing flight mode via GCS works just fine all the time.

[image]

**Version**
+ ArduRover V4.0.0 (0e52bafa)
+ ChibiOS: 0997003f

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[x] Rover
[  ] Submarine

**Airframe type**
Boat, one motor, one ground steering servo

**Hardware type**
+ MatekF405-Win 001F0034

rmackay9, 2021-04-17:
This is really a support question so ideally it should first be reported in the forums.  How do you feel about trying Rover-4.1 and if the problem persists report it here:https://github.com/ArduPilot/ardupilot/issues/17200?

Rover-4.1 can be installed by pushing the "beta firmwares" link on the bottom right of the Install Firmware screen.

If you'd rather stick with stable can you attach an onboard log?

wucke13, 2021-04-18:
> This is really a support question

Why? I'm heaving trouble deciding on this. My feeling is that you deem basically everything a support question? I thought this is very clearly a bug: the indicator in the GUI shows me that I selected a different flight mode than the active one, it's not overridden by anything (as the behavior remains if no GCS is connected), AFAIK there is no condition to block a change to manual mode. I genuinely do not know where there is room for support here?

I will provide an onboard in the coming weeks.

rmackay9, 2021-04-18:
@wucke13,

yes, nearly everything is a support question until it is confirmed to be a software bug by one of the developers.  Enhancement requests don't need to go through the vetting procedure.

changing flight modes using an RC switch is such a commonly used feature (basically every user does this) that it is far more likely to be an issue with the receiver or with the configuration.

I'm going to go ahead and close this but I'll be monitoring this thread in case you post here. Alternatively consider trying Rover-4.1 because I'm very actively monitoring posts in the category linked above.

closed 2021-04-18 by rmackay9