ArduPilot/ardupilot #17206: FENCE_ENABLE prevents arming in non-GPS modes

dakejahl, 2021-04-16:
I just experienced a very puzzling issue with a customer who was trying to arm their copter inside in a non-gps mode. They were getting the prearm error "wating for 3d fix". We tried disabling all the GPS checks and nothing helped. I found this which solved the issue https://discuss.ardupilot.org/t/disabled-pre-arm-checks-but-still-cannot-arm-because-of-no-3d-fix/15559/3

This was very confusing for us because the error was "Waiting for 3D fix" even though we were in a non-GPS mode. I propose
1. a better error message to handle the case for FENCE_ENABLE = 1
2. skip the logic pertaining to FENCE_ENABLE and the corresponding GPS checks when in a non-GPS mode

Thanks!

rmackay9, 2021-04-17:
It's best to raise these types of questions/comments in the discuss forum but in any case, Copter-4.1 has a slight improvement.  It will display a message like below saying that it's the fence that is causing the requirement for the position estimate.

[image]

closed 2021-04-17 by rmackay9

peterbarker, 2021-04-18:
On Fri, 16 Apr 2021, Jacob Dahl wrote:

> This was very confusing for us because the error was "Waiting for 3D fix" even though we were in a non-GPS mode. I propose
>  1. a better error message to handle the case for FENCE_ENABLE = 1

You should have been receiving "Fence requires position" - that's the 
pre-arm failure message fence emits.

Did you not see that?

>  2. skip the logic pertaining to FENCE_ENABLE and the corresponding GPS checks when in a non-GPS mode

I'm afraid this is very much entrenched behaviour now.  People actually 
really still want their fences activated when in manual modes - in fact, 
I'd suggest *mostly* that's when they want them!

One option you might consider is to put fence-enable on a switch.  It does 
move the onus of turning that fence onto the pilot, however, and thatt may 
not suit their workflows.