ArduPilot/ardupilot #17207: Rover: FENCE_ param descriptions shouldn't include Altitude

rmackay9, 2021-04-16:
Rover uses AC_Fence but can't implement the altitude portion so we should change the parameter meta data for Rover so that it doesn't lead people to check the Altitude boxes, etc.  Code is here: https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_Fence/AC_Fence.cpp#L28

rmackay9, 2021-04-21:
resolved in master so closing

closed 2021-04-21 by rmackay9

geofrancis, 2021-09-10:
rather than removing this, would it not be better to replace altitude information with depth information so boats can use it to trigger fence action if the depth gets too deep or shallow.

rmackay9, 2021-09-10:
@geofrancis,

That's a good idea but for now we don't have this feature so better to remove it so it doesn't cause confusion.  If we add a depth failsafe then we can add it back in.