ArduPilot/ardupilot #17215: Copter4.1: cygwin OSError: [Errno 7] Argument list too long

olliw42, 2021-04-18:
I'm compiling ArduCopter4.0.1-beta1 on a Win7 system using cygwin

When I add just one .cpp it fails with error "OSError: [Errno 7] Argument list too long" in the linking stage (full error below).
It seems to hit the ARG_MAX or some other limit.
Proof in the pudding I: When I simply copy the code of said additional .cpp into another already existing .cpp, then it works
Proof in the pudding II: When I simply shorten the names of some existing .cpp then it works
=> solution: probably any method to avoid hitting the limits

FYI

```
$ getconf ARG_MAX
32000

$ ulimit -a
core file size          (blocks, -c) unlimited
data seg size           (kbytes, -d) unlimited
file size               (blocks, -f) unlimited
open files                      (-n) 3200
pipe size            (512 bytes, -p) 8
stack size              (kbytes, -s) 2040
cpu time               (seconds, -t) unlimited
max user processes              (-u) 256
virtual memory          (kbytes, -v) unlimited
```

The complete error

```
Waf: Leaving directory `/cygdrive/c/Users/Olli/Documents/GitHub/ardupilot/build/MatekH743'
Build failed
Traceback (most recent call last):
  File "/cygdrive/c/Users/Olli/Documents/GitHub/ardupilot/modules/waf/waflib/Context.py", line 363, in exec_command
    ret, out, err = Utils.run_process(cmd, kw, cargs)
  File "/cygdrive/c/Users/Olli/Documents/GitHub/ardupilot/modules/waf/waflib/Utils.py", line 978, in run_process
    return run_prefork_process(cmd, kwargs, cargs)
  File "/cygdrive/c/Users/Olli/Documents/GitHub/ardupilot/modules/waf/waflib/Utils.py", line 905, in run_prefork_process
    raise OSError(trace)
OSError: ['/cygdrive/c/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update/bin/arm-none-eabi-ar', 'rcs', 'lib/libArduCopter_libs.a', 'libraries/AP_AccelCal/AP_AccelCal.cpp.0.o', 'libraries/AP_AccelCal/AccelCalibrator.cpp.0.o', 'libraries/AP_ADC/AP_ADC_ADS1115.cpp.0.o', 'libraries/AP_AHRS/AP_AHRS_DCM.cpp.0.o', 'libraries/AP_AHRS/AP_AHRS_Logging.cpp.0.o', 'libraries/AP_AHRS/AP_AHRS_View.cpp.0.o', 'libraries/AP_AHRS/AP_AHRS.cpp.3.o', 'libraries/AP_AHRS/AP_AHRS_NavEKF.cpp.3.o', 'libraries/AP_Airspeed/AP_Airspeed_ASP5033.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_Backend.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_DLVR.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_Health.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_MS4525.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_MS5525.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_MSP.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_SDP3X.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_UAVCAN.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_analog.cpp.0.o', 'libraries/AP_Airspeed/Airspeed_Calibration.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed.cpp.3.o', 'libraries/AP_Airspeed/AP_Airspeed_NMEA.cpp.3.o', 'libraries/AP_Baro/AP_Baro_BMP085.cpp.0.o', 'libraries/AP_Baro/AP_Baro_BMP280.cpp.0.o', 'libraries/AP_Baro/AP_Baro_BMP388.cpp.0.o', 'libraries/AP_Baro/AP_Baro_Backend.cpp.0.o', 'libraries/AP_Baro/AP_Baro_DPS280.cpp.0.o', 'libraries/AP_Baro/AP_Baro_Dummy.cpp.0.o', 'libraries/AP_Baro/AP_Baro_ExternalAHRS.cpp.0.o', 'libraries/AP_Baro/AP_Baro_FBM320.cpp.0.o', 'libraries/AP_Baro/AP_Baro_HIL.cpp.0.o', 'libraries/AP_Baro/AP_Baro_ICM20789.cpp.0.o', 'libraries/AP_Baro/AP_Baro_KellerLD.cpp.0.o', 'libraries/AP_Baro/AP_Baro_LPS2XH.cpp.0.o', 'libraries/AP_Baro/AP_Baro_Logging.cpp.0.o', 'libraries/AP_Baro/AP_Baro_MS5611.cpp.0.o', 'libraries/AP_Baro/AP_Baro_MSP.cpp.0.o', 'libraries/AP_Baro/AP_Baro_SPL06.cpp.0.o', 'libraries/AP_Baro/AP_Baro_UAVCAN.cpp.0.o', 'libraries/AP_Baro/AP_Baro_Wind.cpp.0.o', 'libraries/AP_Baro/AP_Baro.cpp.3.o', 'libraries/AP_Baro/AP_Baro_SITL.cpp.3.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Analog.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_BLHeliESC.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Backend.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Bebop.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_FuelFlow.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_FuelLevel_PWM.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Generator.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Logging.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_Generic.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_NeoDesign.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_Rotoye.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_SUI.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_Solo.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Sum.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_UAVCAN.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor.cpp.3.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Params.cpp.3.o', 'libraries/AP_BoardConfig/IMU_heater.cpp.0.o', 'libraries/AP_BoardConfig/board_drivers.cpp.0.o', 'libraries/AP_BoardConfig/AP_BoardConfig.cpp.3.o', 'libraries/AP_Camera/AP_Camera.cpp.0.o', 'libraries/AP_Camera/AP_Camera_Logging.cpp.0.o', 'libraries/AP_Camera/AP_Camera_SoloGimbal.cpp.0.o', 'libraries/AP_Camera/AP_RunCam.cpp.0.o', 'libraries/AP_CANManager/AP_CANIfaceParams.cpp.0.o', 'libraries/AP_CANManager/AP_CANSensor.cpp.0.o', 'libraries/AP_CANManager/AP_CANTester.cpp.0.o', 'libraries/AP_CANManager/AP_CANTester_KDECAN.cpp.0.o', 'libraries/AP_CANManager/AP_SLCANIface.cpp.0.o', 'libraries/AP_CANManager/AP_CANDriver.cpp.3.o', 'libraries/AP_CANManager/AP_CANManager.cpp.3.o', 'libraries/AP_Common/AP_Common.cpp.0.o', 'libraries/AP_Common/AP_ExpandingArray.cpp.0.o', 'libraries/AP_Common/AP_FWVersion.cpp.0.o', 'libraries/AP_Common/ExpandingString.cpp.0.o', 'libraries/AP_Common/Location.cpp.0.o', 'libraries/AP_Common/NMEA.cpp.0.o', 'libraries/AP_Common/c++.cpp.0.o', 'libraries/AP_Compass/AP_Compass_AK09916.cpp.0.o', 'libraries/AP_Compass/AP_Compass_AK8963.cpp.0.o', 'libraries/AP_Compass/AP_Compass_BMM150.cpp.0.o', 'libraries/AP_Compass/AP_Compass_Calibration.cpp.0.o', 'libraries/AP_Compass/AP_Compass_ExternalAHRS.cpp.0.o', 'libraries/AP_Compass/AP_Compass_HIL.cpp.0.o', 'libraries/AP_Compass/AP_Compass_HMC5843.cpp.0.o', 'libraries/AP_Compass/AP_Compass_IST8308.cpp.0.o', 'libraries/AP_Compass/AP_Compass_IST8310.cpp.0.o', 'libraries/AP_Compass/AP_Compass_LIS3MDL.cpp.0.o', 'libraries/AP_Compass/AP_Compass_LSM303D.cpp.0.o', 'libraries/AP_Compass/AP_Compass_LSM9DS1.cpp.0.o', 'libraries/AP_Compass/AP_Compass_MAG3110.cpp.0.o', 'libraries/AP_Compass/AP_Compass_MMC3416.cpp.0.o', 'libraries/AP_Compass/AP_Compass_MMC5xx3.cpp.0.o', 'libraries/AP_Compass/AP_Compass_MSP.cpp.0.o', 'libraries/AP_Compass/AP_Compass_QMC5883L.cpp.0.o', 'libraries/AP_Compass/AP_Compass_RM3100.cpp.0.o', 'libraries/AP_Compass/AP_Compass_SITL.cpp.0.o', 'libraries/AP_Compass/AP_Compass_UAVCAN.cpp.0.o', 'libraries/AP_Compass/CompassCalibrator.cpp.0.o', 'libraries/AP_Compass/Compass_PerMotor.cpp.0.o', 'libraries/AP_Compass/Compass_learn.cpp.0.o', 'libraries/AP_Compass/AP_Compass.cpp.3.o', 'libraries/AP_Compass/AP_Compass_Backend.cpp.3.o', 'libraries/AP_Declination/AP_Declination.cpp.0.o', 'libraries/AP_Declination/tables.cpp.0.o', 'libraries/AP_GPS/AP_GPS.cpp.0.o', 'libraries/AP_GPS/AP_GPS_ERB.cpp.0.o', 'libraries/AP_GPS/AP_GPS_ExternalAHRS.cpp.0.o', 'libraries/AP_GPS/AP_GPS_GSOF.cpp.0.o', 'libraries/AP_GPS/AP_GPS_MAV.cpp.0.o', 'libraries/AP_GPS/AP_GPS_MSP.cpp.0.o', 'libraries/AP_GPS/AP_GPS_MTK.cpp.0.o', 'libraries/AP_GPS/AP_GPS_MTK19.cpp.0.o', 'libraries/AP_GPS/AP_GPS_NMEA.cpp.0.o', 'libraries/AP_GPS/AP_GPS_NOVA.cpp.0.o', 'libraries/AP_GPS/AP_GPS_SBF.cpp.0.o', 'libraries/AP_GPS/AP_GPS_SBP.cpp.0.o', 'libraries/AP_GPS/AP_GPS_SBP2.cpp.0.o', 'libraries/AP_GPS/AP_GPS_SIRF.cpp.0.o', 'libraries/AP_GPS/AP_GPS_UAVCAN.cpp.0.o', 'libraries/AP_GPS/AP_GPS_UBLOX.cpp.0.o', 'libraries/AP_GPS/GPS_Backend.cpp.0.o', 'libraries/AP_GPS/MovingBase.cpp.0.o', 'libraries/AP_GPS/RTCM3_Parser.cpp.0.o', 'libraries/AP_HAL/CANIface.cpp.0.o', 'libraries/AP_HAL/DSP.cpp.0.o', 'libraries/AP_HAL/Device.cpp.0.o', 'libraries/AP_HAL/EventHandle.cpp.0.o', 'libraries/AP_HAL/GPIO.cpp.0.o', 'libraries/AP_HAL/HAL.cpp.0.o', 'libraries/AP_HAL/RCOutput.cpp.0.o', 'libraries/AP_HAL/Scheduler.cpp.0.o', 'libraries/AP_HAL/Semaphores.cpp.0.o', 'libraries/AP_HAL/Storage.cpp.0.o', 'libraries/AP_HAL/Util.cpp.0.o', 'libraries/AP_HAL/system.cpp.0.o', 'libraries/AP_HAL/utility/BetterStream.cpp.0.o', 'libraries/AP_HAL/utility/RCOutput_Tap.cpp.0.o', 'libraries/AP_HAL/utility/RCOutput_Tap_Linux.cpp.0.o', 'libraries/AP_HAL/utility/RingBuffer.cpp.0.o', 'libraries/AP_HAL/utility/Socket.cpp.0.o', 'libraries/AP_HAL/utility [... cut ...] odules/uavcan/libuavcan/src/uc_error.cpp.0.o']
Traceback (most recent call last):
  File "<string>", line 33, in run
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 7] Argument list too long: '/cygdrive/c/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update/bin/arm-none-eabi-ar'
```

rmackay9, 2021-04-18:
This came up for me while I was testing PR https://github.com/ArduPilot/ardupilot/pull/16725.  Tridge says, "i think we'll need to break up the main library build into multiple components to reduce command line length".

Thanks for the report.

rmackay9, 2021-04-20:
Discussed on the dev call today and it's probably this line that is the issue: https://github.com/ArduPilot/waf/blob/master/waflib/Tools/ccroot.py#L225

tridge, 2021-04-20:
@rmackay9 can you please send me the exact output you get on the failure for CubeOrange with this PR

rmackay9, 2021-04-20:
[730/788] Linking build/CubeOrange/lib/libArduCopter_libs.a
[731/788] Compiling ArduCopter/sensors.cpp
[732/788] Compiling ArduCopter/mode_loiter.cpp
Waf: Leaving directory `/cygdrive/c/Users/rmackay9/Documents/GitHub/r9-ardupilot/build/CubeOrange'
Build failed
Traceback (most recent call last):
File "/cygdrive/c/Users/rmackay9/Documents/GitHub/r9-ardupilot/modules/waf/waflib/Task.py", line 338, in process
ret = self.run()
File "/cygdrive/c/Users/rmackay9/Documents/GitHub/r9-ardupilot/modules/waf/waflib/Tools/ccroot.py", line 237, in wrap
return old(self)
File "", line 22, in f
File "/cygdrive/c/Users/rmackay9/Documents/GitHub/r9-ardupilot/modules/waf/waflib/Tools/ccroot.py", line 179, in exec_command
ret = super(link_task, self).exec_command(*k, **kw)
File "/cygdrive/c/Users/rmackay9/Documents/GitHub/r9-ardupilot/modules/waf/waflib/Task.py", line 321, in exec_command
return self.generator.bld.exec_command(cmd, **kw)
File "/cygdrive/c/Users/rmackay9/Documents/GitHub/r9-ardupilot/modules/waf/waflib/Context.py", line 365, in exec_command
raise Errors.WafError('Execution failure: %s' % str(e), ex=e)
WafError: Execution failure: ['/cygdrive/c/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update/bin/arm-none-eabi-ar', 'rcs', 'lib/libArduCopter_libs.a', 'libraries/AP_AccelCal/AP_AccelCal.cpp.0.o', 'libraries/AP_AccelCal/AccelCalibrator.cpp.0.o', 'libraries/AP_ADC/AP_ADC_ADS1115.cpp.0.o', 'libraries/AP_AHRS/AP_AHRS_DCM.cpp.0.o', 'libraries/AP_AHRS/AP_AHRS_Logging.cpp.0.o', 'libraries/AP_AHRS/AP_AHRS_View.cpp.0.o', 'libraries/AP_AHRS/AP_AHRS.cpp.3.o', 'libraries/AP_AHRS/AP_AHRS_NavEKF.cpp.3.o', 'libraries/AP_Airspeed/AP_Airspeed_ASP5033.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_Backend.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_DLVR.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_Health.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_MS4525.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_MS5525.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_MSP.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_SDP3X.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_UAVCAN.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed_analog.cpp.0.o', 'libraries/AP_Airspeed/Airspeed_Calibration.cpp.0.o', 'libraries/AP_Airspeed/AP_Airspeed.cpp.3.o', 'libraries/AP_Airspeed/AP_Airspeed_NMEA.cpp.3.o', 'libraries/AP_Baro/AP_Baro_BMP085.cpp.0.o', 'libraries/AP_Baro/AP_Baro_BMP280.cpp.0.o', 'libraries/AP_Baro/AP_Baro_BMP388.cpp.0.o', 'libraries/AP_Baro/AP_Baro_Backend.cpp.0.o', 'libraries/AP_Baro/AP_Baro_DPS280.cpp.0.o', 'libraries/AP_Baro/AP_Baro_Dummy.cpp.0.o', 'libraries/AP_Baro/AP_Baro_ExternalAHRS.cpp.0.o', 'libraries/AP_Baro/AP_Baro_FBM320.cpp.0.o', 'libraries/AP_Baro/AP_Baro_HIL.cpp.0.o', 'libraries/AP_Baro/AP_Baro_ICM20789.cpp.0.o', 'libraries/AP_Baro/AP_Baro_KellerLD.cpp.0.o', 'libraries/AP_Baro/AP_Baro_LPS2XH.cpp.0.o', 'libraries/AP_Baro/AP_Baro_Logging.cpp.0.o', 'libraries/AP_Baro/AP_Baro_MS5611.cpp.0.o', 'libraries/AP_Baro/AP_Baro_MSP.cpp.0.o', 'libraries/AP_Baro/AP_Baro_SPL06.cpp.0.o', 'libraries/AP_Baro/AP_Baro_UAVCAN.cpp.0.o', 'libraries/AP_Baro/AP_Baro_Wind.cpp.0.o', 'libraries/AP_Baro/AP_Baro.cpp.3.o', 'libraries/AP_Baro/AP_Baro_SITL.cpp.3.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Analog.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_BLHeliESC.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Backend.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Bebop.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_FuelFlow.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_FuelLevel_PWM.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Generator.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Logging.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_Generic.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_NeoDesign.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_Rotoye.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_SUI.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_SMBus_Solo.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Sum.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor_UAVCAN.cpp.0.o', 'libraries/AP_BattMonitor/AP_BattMonitor.cpp.3.o', 'libraries/AP_BattMonitor/AP_BattMonitor_Params.cpp.3.o', 'libraries/AP_BoardConfig/IMU_heater.cpp.0.o', 'libraries/AP_BoardConfig/board_drivers.cpp.0.o', 'libraries/AP_BoardConfig/AP_BoardConfig.cpp.3.o', 'libraries/AP_Camera/AP_Camera.cpp.0.o', 'libraries/AP_Camera/AP_Camera_Logging.cpp.0.o', 'libraries/AP_Camera/AP_Camera_SoloGimbal.cpp.0.o', 'libraries/AP_Camera/AP_RunCam.cpp.0.o', 'libraries/AP_CANManager/AP_CANIfaceParams.cpp.0.o', 'libraries/AP_CANManager/AP_CANSensor.cpp.0.o', 'libraries/AP_CANManager/AP_CANTester.cpp.0.o', 'libraries/AP_CANManager/AP_CANTester_KDECAN.cpp.0.o', 'libraries/AP_CANManager/AP_SLCANIface.cpp.0.o', 'libraries/AP_CANManager/AP_CANDriver.cpp.3.o', 'libraries/AP_CANManager/AP_CANManager.cpp.3.o', 'libraries/AP_Common/AP_Common.cpp.0.o', 'libraries/AP_Common/AP_ExpandingArray.cpp.0.o', 'libraries/AP_Common/AP_FWVersion.cpp.0.o', 'libraries/AP_Common/ExpandingString.cpp.0.o', 'libraries/AP_Common/Location.cpp.0.o', 'libraries/AP_Common/NMEA.cpp.0.o', 'libraries/AP_Common/c++.cpp.0.o', 'libraries/AP_Compass/AP_Compass_AK09916.cpp.0.o', 'libraries/AP_Compass/AP_Compass_AK8963.cpp.0.o', 'libraries/AP_Compass/AP_Compass_BMM150.cpp.0.o', 'libraries/AP_Compass/AP_Compass_Calibration.cpp.0.o', 'libraries/AP_Compass/AP_Compass_ExternalAHRS.cpp.0.o', 'libraries/AP_Compass/AP_Com [... cut ...] .0.o', 'libraries/AC_Avoidance/AP_OAVisGraph.cpp.0.o', 'libraries/AC_Avoidance/AC_Avoid.cpp.3.o', 'libraries/AC_Avoidance/AP_OABendyRuler.cpp.3.o', 'libraries/AC_Avoidance/AP_OADatabase.cpp.3.o', 'libraries/AP_LandingGear/AP_LandingGear.cpp.0.o', 'libraries/AP_RobotisServo/AP_RobotisServo.cpp.0.o', 'libraries/AP_ToshibaCAN/AP_ToshibaCAN.cpp.0.o', 'libraries/AP_NMEA_Output/AP_NMEA_Output.cpp.0.o', 'libraries/AP_Filesystem/AP_Filesystem.cpp.0.o', 'libraries/AP_Filesystem/AP_Filesystem_FATFS.cpp.0.

closed 2021-04-20 by tridge