ArduPilot/ardupilot #17224: Uart delay start when power on

Abner0907, 2021-04-19:
Hi, I am not sure this is only my issues. I am very appreciate that some one can look the details. 

I have the text that:
I add a power button in my private hardware. When I push the button, the system is power on and start working(_Push HIGH my a gpio pin_), everything is fine. But When I look at the `AP_SerialManager` thread.  The system can get in the code below after 5 seconds. I can not catch the reason. 
[image]
**BTW**
1: My hardware is modified but similar as the pixhawk V4.
2: _The code call/initialize in the [init_ardupilot](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/system.cpp#L16) function without delay._

magicrub, 2021-04-19:
I don't think I understand the question

Abner0907, 2021-04-19:
@magicrub  The uart start to work  later 5 seconds  than the bootloader is started or system power plug in?

andyp1per, 2021-04-19:
There is always a 5 second delay at boot by design

Abner0907, 2021-04-19:
>There is always a 5 second delay at boot by design

@andyp1per  Hi, is this hide in the bootloader? If so, I will review the bootloader code

WickedShell, 2021-05-18:
> > There is always a 5 second delay at boot by design
> 
> @andyp1per Hi, is this hide in the bootloader? If so, I will review the bootloader code

Yes, this is by design and inside the bootloader. I'm closing as this appears to be a support issue best asked on discord or the forums.

closed 2021-05-18 by WickedShell