ArduPilot/ardupilot #17227: Allow vehicle-to-vehicle photography with `SYSID_TARGET`

arikrupnik, 2021-04-20:
## Feature request

Antenna tracker follows a vehicle, pointing antenna or camera at it. It would be useful to allow other vehicle types to perform the same function. The most immediate use case is a "camera trolley," a rover that travels to a specific vantage point to video the target vehicle from it. Ultimately, this feature would allow air-to-air photography and--for vehicles that follow other vehicles--chase camera footage.

**Describe alternatives you've considered**

Putting two AP boards on the same vehicle, one to control the vehicle, another to control the tracker.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[ x ] Rover
[ x ] Submarine

**Additional context**

Some discussion on this has occurred in Antenna Tracker discord channel. Cc @auturgy

peterbarker, 2021-04-20:
On Mon, 19 Apr 2021, arikrupnik wrote:

> Antenna tracker follows a vehicle, pointing antenna or camera at it. It would be useful to allow other vehicle types to perform the same function. The most immediate use case is a "camera trolley," [...]

AP_Mount already has the option to track SYSID_TARGET.

rmackay9, 2021-04-20:
Great, I think I'll close this issue then.  We probably need to document the feature though..

closed 2021-04-20 by rmackay9

arikrupnik, 2021-04-20:
I thought that was possible, but the documentation tripped me up :=) Thank you @rmackay9