ArduPilot/ardupilot #17243: Orange cube telemetry drops more packets than Black cube

TorbjornHouge, 2021-04-21:
## Bug report

**Issue details**
The problem has been observed in a fixed wing UAV in the field, and on a desktop setup using USB and the telemetry outputs. It seems that even connecting just the cube to the computer using USB recreates the problem. The telemetry streams are then received by MavProxy, in order to merge and connect to a single instance of Mission Planner.

The problem is that while the Black Cube achieves a 100 % telemetry throughput, and 0 dropped packets in MavProxy, the same firmware (tested with several, including 4.0.9) and config on the Orange Cube gives a throughput of 96 to 99 %. When using dual links that both go to the same MavProxy, it seems that whatever is called Link 2 is down to 96 %, also using hardwired connection. When using radios in the field, this is further reduced, while the Black Cube keeps better.

**Version**
4.0.4 and later (not present in 4.0.3)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Fixed wing

**Hardware type**
Pixhawk Orange Cube

**Logs**
Described with screenshots here: https://discuss.ardupilot.org/t/orange-cube-telemetry-drops-more-packets-than-black-cube/68187/7

rmackay9, 2021-04-22:
My random guess is that it is because there are 3 IMUs enabled on the CubeOrange and this is eating more telemetry bandwidth.  You could test this by reducing EKx_IMU_MASK to 3 or 1 (instead of 7).

In general issues should be raised in the forums before being raised here on the to-do list which is really for confirmed bugs and enhancements.  Still, this is Plane so I'll leave it for the plane guys to decide.

TorbjornHouge, 2021-04-22:
Still strange since it seems to occur on whatever is the second link on MavProxy. In testing, that means either Telem1 or USB connection, not really mattering which it is at the time. Also have tested with baudrates set up towards maximum of what I managed to get the system to handle. Sorry if I have bypassed a confirmed bug procedure, but there has been no response to this issue in the forum for 1,5 month.

IamPete1, 2021-08-09:
This should have been resolved by the change to use the FIFO, https://github.com/ArduPilot/ardupilot/pull/17959

H7 is also less tolerant of baudrate errors than F4 due to the higher clock-rate. If there are still issues it might be worth checking the thing your connecting to is hitting the baudrate it is supposed to be. This is often the issue if using none-usb connections to companion computers. 

@TorbjornHouge It would be great if you could check if the issue is resolved on 4.1.dev now we are using the FIFO.